rtklibexplorer/RTKLIB #14: RTKCONV regression

GPSFan, 2017-09-02:
The most recent commit: 01271608c1a6a5de87eb1d64278ac0d936dd2129
modifies src/rev/ublox.c

-        sys==SYS_CMP?CODE_L1I:(sys==SYS_GAL?CODE_L1C:CODE_L1C);
+        sys==SYS_CMP?CODE_L1I:(sys==SYS_GAL?CODE_L1X:CODE_L1C);
         
Which reverts part of this commit: 979b8581ebfba3a51eb1bd414bd702a80af26899

            sys==SYS_CMP?CODE_L1I:(sys==SYS_GAL?CODE_L1X:CODE_L1C);
+        sys==SYS_CMP?CODE_L1I:(sys==SYS_GAL?CODE_L1C:CODE_L1C);

The end result is that at least rtkconv will not convert .ubx Galileo data to Rinex.

rtklibexplorer, 2017-09-03:
This should be OK.  My original change from L1X to L1C in ublox.c made Galileo work but was not really correct since the code type was not actually L1C.  A couple changes back I added L1X to the list of valid codes in convrnx.c for ublox so the original version of ublox.c should now work.  I just tested it again with RTKCONV and it looks OK.  Is it possible you did not have Galileo selected in the options menu when you ran RTKCONV?

GPSFan, 2017-09-03:
That must have been the problem, I could have sworn that I verified that it was checked before I posted this issue. Sorry for the noise. I just re-ran my .ubx through rtkconv and it produced the expected Galileo data. I've been trying to track down an issue I'm seeing with the same data & corrections producing results that differ by 5cm or so depending on variations in parameters. I could understand 20cm or larger being due false AR fix. Unfortunately I don't have a benchmark anywhere close, and my nearest CORS is 25+ km away. Sitting on a point I get +/- 1cm variations, but run the same data again with slightly different parameters and the whole dataset is off by 2-7 cm. Maybe it's just toooo long a baseline. Anyway thanks for the help.

closed 2017-09-04 by GPSFan

rtklibexplorer, 2017-09-06:
Errors will increase with baseline, 25+ km is a relatively long baseline
and your errors may not be that unreasonable given that distance.  You can
reduce errors in long baseline solutions somewhat by using precise
ephemeris and ionoospheric data.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>