rtklibexplorer/RTKLIB #17: How to compile and run the QT part on Linux?

matmis, 2017-12-17:
I don't know much about qt, but when I run the following:

mkdir build
cd build
qmake ../app/rtknavi_qt/rtknavi_qt.pro 
make

I got the error:

```
../app/appcmn_qt/serioptdlg.cpp:13:50: fatal error: QtExtSerialPort/qextserialenumerator.h: No such file or directory
 #include <QtExtSerialPort/qextserialenumerator.h>
compilation terminated.
Makefile:231: recipe for target 'qextserialenumerator.o' failed
make: *** [qextserialenumerator.o] Error 1

```

Any suggestions?

DavidKelleySCSC, 2017-12-17:
Two quick questions.   First are you using the right repository, or do you want **cartoocm/RTKLIB-QT** for this?  Second, that error normally occurs when you are using Qt rev 4.x and fail to include the 3rd party serial stuff (it is now built into rev 5.x).  [I do not know what version JensReimann used] You are better off going to Qt 5.x  (currently at 5.9.1) if you can. Finally, as Qt is more of less a GUI development environment, many folks (most?) click on the *.pro file, pick the build tool chain you want (called _kits_ now), and go on - rather than a command line.

matmis, 2017-12-17:
Thank you for you answer!

I think that I'm in the right repository because I use U-blox receivers.
I installed `qt5-default` and did the same thing.
Now when I run 
`qmake ../app/rtknavi_qt/rtknavi_qt.pro`
the error is:

```
Info: creating stash file /tmp/RTKLIB/build/.qmake.stash
Project ERROR: Unknown module(s) in QT: serialport
```

libqt5serialport5 is installed. (if that's useful to know)
`libqt5serialport5/zesty,now 5.7.1~20161021-2build1~1 amd64 [installed,automatic]`

Jingzhe-Gao, 2018-10-15:
I

> Thank you for you answer!
> 
> I think that I'm in the right repository because I use U-blox receivers.
> I installed `qt5-default` and did the same thing.
> Now when I run
> `qmake ../app/rtknavi_qt/rtknavi_qt.pro`
> the error is:
> 
> ```
> Info: creating stash file /tmp/RTKLIB/build/.qmake.stash
> Project ERROR: Unknown module(s) in QT: serialport
> ```
> 
> libqt5serialport5 is installed. (if that's useful to know)
> `libqt5serialport5/zesty,now 5.7.1~20161021-2build1~1 amd64 [installed,automatic]`
> 
> **UPDATE**
> I installed `libqt5serialport5-dev` and I could generate the "Makefile" with qmake
> but when I run `make`
> I got the following error:
> 
> `make: *** No rule to make target '../../src/libRTKLib.a', needed by 'rtknavi_qt'. Stop.`

I came across the same problem, so have you solved this?

yves84, 2018-10-16:
## Workaround
ur need to compile the static lib libRTKLib.a:
```
cd <PATH_TO_RTKLIB_ROOT_DIR>
qmake RTKLib.pro
make
```
The build process will stopped with the same error:
```
ar cqs libRTKLib.a rtkcmn.o convkml.o convrnx.o convgpx.o datum.o download.o ephemeris.o geoid.o gis.o ionex.o lambda.o options.o pntpos.o postpos.o ppp.o ppp_ar.o ppp_corr.o preceph.o qzslex.o rcvraw.o rinex.o rtcm.o rtcm2.o rtcm3.o rtcm3e.o rtkpos.o rtksvr.o sbas.o solution.o stream.o streamsvr.o tides.o tle.o binex.o crescent.o gw10.o javad.o novatel.o nvs.o rcvlex.o rt17.o septentrio.o skytraq.o ss2.o ublox.o tersus.o comnav.o swiftnav.o cmr.o
make[1]: Verzeichnis „/home/yves/Apps/RTKLIB/src“ wird verlassen
cd app/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /home/yves/Apps/RTKLIB/app/app.pro ) && make -f Makefile 
make[1]: Verzeichnis „/home/yves/Apps/RTKLIB/app“ wird betreten
make[1]: *** Keine Regel vorhanden, um das Ziel „../../src/libRTKLib.a“, 
  benötigt von „rtknavi_qt“, zu erstellen.  Schluss.
```
Befor the error comes, it has create the `libRTKLib.a` and u can find it in the `src` dir:
```
$: find . -iname '*RTKLib.a*'
./src/libRTKLib.a
```
The build process stopped while building the `app.pro` in the `app` dir and try to link (?) the `libRTKLib.a` in `../../libRTKLib.a` what is wrong (there is in `../src/`).
U can go into the rtknav_qt dir and build it because there is a .pro file too:
```
cd app/rtknavi_qt
qmake rtknavi_qt.pro
make
```

Sry, i dont use QMake or QTCreator and dont know the hole build processes. but it works for me.

Cyrille37, 2019-02-11:
Thanks for the solution
```
qmake *.pro
make
```

Cyrille37, 2019-02-11:
Next error...

`rtknavi_qt` compilation fails in `mondlg.cpp` because of incompatible call for function `code2obs()` (_in rtklib.h_) : 
```
mondlg.cpp:1864:38: error: too few arguments to function ‘char* code2obs(int, unsigned char, int*)’
In file included from mondlg.cpp:7:0:
../../src/rtklib.h:1453:14: note: declared here
 EXPORT char *code2obs(int sys, unsigned char code, int *freq);
              ^~~~~~~~
Makefile:884: recipe for target 'mondlg.o' failed
```

Cyrille37, 2019-02-11:
As a hack I modified `code2base` call with add `0` as first missing parameter.
```
line 1007:
code=code2obs(0,obs[i].code[k],NULL);
line 1864:
char *code=code2obs(0, i-17,NULL);
```

Cyrille37, 2019-02-11:
Next error:
`rtknavi_qt` compilation fails because missing `libgd`

solution (easy):
```
sudo apt-get install libgd-dev
```

sonicpp, 2021-08-22:
As the main RTKLib.pro has been deleted, one must compile library and apps this way:

```
git clone https://github.com/rtklibexplorer/RTKLIB.git
cd RTKLIB/
```

```
pushd .
cd src
qmake && make
popd
```

```
cd app/qtapp
qmake && make
```

HamishB, 2021-12-10:
Hi, I'm getting an error trying to build the RTKLIB-demo5 qt apps on Ubuntu 18.04 LTS,

```
$ qmake && make
[...]
cd rtknavi_qt/ && make -f Makefile 
make[1]: Entering directory '[...]/RTKLIB-demo5/app/qtapp/rtknavi_qt'
g++ -c -m64 -pipe -g -Wall -W -D_REENTRANT -D_RTLDLL -DNO_STRICT -DTRACE -DENAGLO -DENAQZS -DENAGAL -DENACMP -DENAIRN -DNFREQ=5 -DNEXOBS=3 -DSVR_REUSEADDR -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I../../../src -I../appcmn_qt -I. -I. -o serioptdlg.o ../appcmn_qt/serioptdlg.cpp
../appcmn_qt/serioptdlg.cpp:12:10: fatal error: QSerialPortInfo: No such file or directory
 #include <QSerialPortInfo>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
```
the QSerialPortInfo header is installed at `/usr/include/x86_64-linux-gnu/qt5/QtSerialPort/QSerialPortInfo` by the `libqt5serialport5-dev` package. Qt5 version 5.9.5.

any ideas?

thanks,
Hamish

gaofenzi, 2021-12-10:
您好，辛苦啦！梦寒已经收到您的珍贵来件，会在最短的时间内给予您最想要的答复！

HamishB, 2021-12-11:
ah, the rtk*_qt apps' Makefiles are all written including qt4, but the source code uses qt5 functions. The system's qmake is using Qt4 as linked by qtchooser?

HamishB, 2021-12-11:
Ok, got it built, I needed to install the qt5-default package as per
  https://rtklibexplorer.wordpress.com/2020/12/18/building-rtklib-code-in-linux/

rtklibexplorer, 2023-04-17:
Qt apps are no longer supported in the 2.4.3 and demo5 versions of RTKLIB

closed 2023-04-17 by rtklibexplorer