rtklibexplorer/RTKLIB #21: rtkrcv segfault on auto-start

nathanja, 2018-03-26:
When adding the "-s" flag to rtkrcv, if an options file has been specified that sets outputstr1/2 or logstr1/2 to a file type, the binary segfaults on start.

If the binary is loaded with the options file and then manually started, there is no issue.

The same issue does not appear on rtkrcv compiled using RTKLIB 2.4.2 or 2.4.3 source

In this instance, rtkrcv was compiled using make on ubuntu16.04

```
./rtkrcv -s -o rtkexplorer_custom.conf
Segmentation fault (core dumped)
```

sezeryalcin, 2018-09-04:
This is a known issue since b26 but no fix yet.
I have a patch just to bypass it which forces autostart without -s flag

rtklibexplorer, 2018-12-18:
Fixed with the 11/23/18 commit

closed 2018-12-18 by rtklibexplorer