rtklibexplorer/RTKLIB #45: When was tadj (time adjust) added to gen_obs_glo() call?

DavidKelleySCSC, 2019-09-30:
Not an issue, just trying to keep one of our clone copies aligned.  Trying to get a feel for how far back the two diverged, so I can seek for related changes.  **Can anyone recall when this was added?** 

What first brought this issue up was a failure in the b31a release (using the strsrv tool) to translate GLO MT1012 to MT1010 where it appears that the carrier frequency is always getting set to -1 (err),  The codes fail for the translation is MT1012->MT1012 as well.  The old code also has an issue where lambda is not always computed that seems to be dealt with as well in the new. 

Running this with b33a, and the same settings, the translate of RTCM3 to RTCM3, GLO MT1012  to 1010/1011/1012 is working fine.

rtklibexplorer, 2019-09-30:
Tadj was added to gen_obs_glo() on 2/2/17 and first appeared in version b26a but I don't believe this is related to the issues you describe.  I believe these were fixed by the commit on 9/17/19 which first appeared in the current b33a release.

DavidKelleySCSC, 2019-09-30:
Thanks of this and the quick reply.  It gives me an anchor in time to be looking for other changes.  I managed to merge the current rtcm3e.c into my project without too many issues.  And it is correctly encoding MT1009~1012 now.  There is a new 'freq' term added to the obs struct I will need to look over carefully as part of this, our branch is quite old.  We have heavily modded rtklib.h with our own elements so this sort of thing so there is always a fear of introduced conflicts at out end.

closed 2019-10-24 by DavidKelleySCSC