rtklibexplorer/RTKLIB #48: RTK-Post Windows 10 GUI buttons cutoff, only5% visible.

lidar532, 2019-11-27:
I got the following screen from the b33b version of rtk-post.  All the earlier version have been fine.  

W. Wright

[image]

rtklibexplorer, 2019-11-27:
I can't duplicate what you are seeing on my computer.  Are you launching
RTKPOST directly or from RTKLAUNCH?  The b33b version of RTKLAUNCH defaults
to running the 64 bit executables for the first time.  The previous
versions all defaulted to running the 32 bit versions. I don't know why
this would cause the issue you are seeing but I suspect it may
have something to do with it.  The window size is contained in the .ini
file and if there is no .ini file it defaults to what should be a
reasonable window size.  The 32 bit version (RTKPOST) has a separate .ini
file from the 64 bit version (RTKPOST_WIN64).  One thing you might try is
copying the rtkpost.ini file from an older executable folder into the b33b
executable folder and renaming it rtkpost_win64.ini.

If you are using RTKLAUNCH and prefer to stick to the 32 bit exectuables
you can change the options by clicking on the small square near the top
right corner.

Let me know what you find so if anyone else runs into this problem, I can
let them know what worked for you.

Best regards,

Tim

anathan, 2019-11-28:
This is likely your window scaling settings. Try adjusting that in control panel (https://www.lifewire.com/using-windows-10-display-scaling-4587328) to 100%

lidar532, 2019-12-07:
The cutoff buttons were indeed caused by the windows-10 scaling settings as suggested by Anathan.  I adjusted to 100%, and the problem went away.  FYI, the 32bit version worked correctly.  It was only the 64bit version that was affected by the scaling.

Best regards,
Wayne

closed 2020-12-19 by rtklibexplorer