rtklibexplorer/RTKLIB #67: bug in reading precise ephemeris (number of satellites)

freeExec, 2020-09-29:
I think the arguments are mixed up here:
https://github.com/rtklibexplorer/RTKLIB/blob/demo5/src/preceph.c#L83
`ns=(int)str2num(buff,4,2);`

The third line of the SP3-file contains the text: number of satellites
`+__114   C06C07C08C09C10C11C12C13C14C16C19C20C21C22C23C24C25`

It reads two characters starting from position 4, gets 14. But I must on the contrary to get 114.
Now he reads the first 14 entries for each epoch, and these turn out to be BeiDou. Accordingly, nothing further works without GPS.

shegnauer, 2020-10-04:
As I understand the documentation, chapter E.4, only SP3-c files are supported, i.e. SP3 version 'c'. The newer SP3-d files which allow >85 satellites would require a rewrite/update of the respective functions.

YanqingHou, 2020-10-08:
The sp3-c to sp3-d modification has been made here: https://github.com/tomojitakasu/RTKLIB/pull/529

rtklibexplorer, 2020-11-21:
Added 2.4.3 PR #529 into 11/21/20 commit

closed 2020-11-21 by rtklibexplorer