rtklibexplorer/RTKLIB #73: error: too few arguments to function ‘void uniqnav(nav_t*, int)’

jbeilin, 2020-11-26:
Hello, 

I get an error when I try to build RTKLIB/demo5 

plotdata.cpp: In member function ‘int Plot::ReadObsRnx(const QStringList&, obs_t*, nav_t*, sta_t*)’:
plotdata.cpp:264:16: error: too few arguments to function ‘void uniqnav(nav_t*, int)’
  264 |     uniqnav(nav);
      |                ^
In file included from plotdata.cpp:14:
../../src/rtklib.h:1543:13: note: declared here
 1543 | EXPORT void uniqnav(nav_t *nav, int nf);

jbeilin, 2020-11-28:
I managed to build the lib and programs after a couple of changes : 
rtklib.h
1/ I had to add theses lines at line 278
#define FREQTYPE_L6 0x08                /* frequency type: E6/LEX/B3 */
#define FREQTYPE_L7 0x10                /* frequency type: E5b/B2 */
#define FREQTYPE_L8 0x20                /* frequency type: E5(a+b) */
#define FREQTYPE_L9 0x40                /* frequency type: S */

2/ plotdata.cpp
line 264 uniqnav(nav); -> uniqnav(nav, 2);
line 297 uniqnav(&Nav); -> uniqnav(&Nav, 2);
Not sure for the value "2"... 

It seems that there is a version mismatch between the lib and the applications. 

Did I made something wrong ?

rtklibexplorer, 2020-11-28:
Sorry, it was my mistake in the previous commit.  It should be OK now with yesterday's commit.

jbeilin, 2020-11-28:
I still get theses errors : 
codeopt.cpp:333:49: error: ‘FREQTYPE_L7’ was not declared in this scope; did you mean ‘FREQTYPE_L1’?
  333 |     E27->setEnabled((NavSys&SYS_GAL)&&(FreqType&FREQTYPE_L7));
      |                                                 ^~~~~~~~~~~
      |                                                 FREQTYPE_L1
codeopt.cpp:336:49: error: ‘FREQTYPE_L6’ was not declared in this scope; did you mean ‘FREQTYPE_L1’?
  336 |     E30->setEnabled((NavSys&SYS_GAL)&&(FreqType&FREQTYPE_L6));
      |                                                 ^~~~~~~~~~~
      |                                                 FREQTYPE_L1
codeopt.cpp:341:49: error: ‘FREQTYPE_L8’ was not declared in this scope; did you mean ‘FREQTYPE_L1’?
  341 |     E37->setEnabled((NavSys&SYS_GAL)&&(FreqType&FREQTYPE_L8));
      |                                                 ^~~~~~~~~~~
      |                                                 FREQTYPE_L1
codeopt.cpp:371:49: error: ‘FREQTYPE_L9’ was not declared in this scope; did you mean ‘FREQTYPE_L1’?
  371 |     I52->setEnabled((NavSys&SYS_CMP)&&(FreqType&FREQTYPE_L9));
      |                                                 ^~~~~~~~~~~
      |                                                 FREQTYPE_L1

Adding  theses lines in rtklib.h at line 278 corrects this issue
#define FREQTYPE_L6 0x08 /* frequency type: E6/LEX/B3 /
#define FREQTYPE_L7 0x10 / frequency type: E5b/B2 /
#define FREQTYPE_L8 0x20 / frequency type: E5(a+b) /
#define FREQTYPE_L9 0x40 / frequency type: S */

rtklibexplorer, 2020-11-28:
I pulled the linux QT GUI code into the demo5 RTKLIB code from the QT branch of RTKLIB at one point a few years ago but have not been supporting this variant on a regular basis.  If anybody can provide pull requests to keep these builds current I would be happy to pull them in as long as they are benign to the other builds.  The errors you list can be addressed by porting changes from rtkconv/codeopt.cpp into rtkconv_qt/codeopt.cpp. 

The demo5 code does support all of the command line versions of the linux apps, so your other option is to build and run CONVBIN.

rtklibexplorer, 2020-12-19:
The errors in the linux Qt builds are fixed in the most recent code.

closed 2020-12-19 by rtklibexplorer