rtklibexplorer/RTKLIB #81: Typo on variable name

jcoelho93, 2021-03-31:
The [latest changes](https://github.com/rtklibexplorer/RTKLIB/commit/0f78ce4cfc67c048aada8736a6b1b99324330eed#diff-af224220568ede26320e079eddcfc17043e09c2249d45375c6daf25cfdd1175fR1698), I believe, have introduced a small typo in a variable name in `src/rtkpos.c`.

When I run `make all` I get the following error:
```
../../../../src/rtkpos.c: In function 'resamb_LAMBDA':
../../../../src/rtkpos.c:1698:79: error: 'Qbb' undeclared (first use in this function); did you mean 'QQb'?
 1698 |     free(y); free(DP); free(b); free(db); free(Qb); free(Qab); free(QQ); free(Qbb);
      |                                                                               ^~~
      |                                                                               QQb
../../../../src/rtkpos.c:1698:79: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [makefile:30: rtkpos.o] Error 1
```

aberenyi, 2021-03-31:
Thanks for reporting this @jcoelho93, I was just about to do the same!

rtklibexplorer, 2021-04-01:
Thanks - I've fixed the compile issue with the most recent commit

closed 2021-04-01 by rtklibexplorer