rtklibexplorer/RTKLIB #83: postpos Process only single external event mark between data epoch

a1ien, 2021-04-05:
If we have multiple external event `postpos` take only first and produce mark for first only.
For example part of obs file
```
...
> 2021 03 18 12 19 04.5950000  0 22
...
R02  22864724.483   122010555.050        2730.966          41.000
> 2021  3 18 12 19  4.6232250  5  0
> 2021  3 18 12 19  4.6273160  5  0
> 2021 03 18 12 19 04.6950000  0 22
G32  21946389.318   115329062.952       -2587.341          41.000    21946381.175    89866772.941       -2016.021          38.000
```
Produce 
```
% (lat/lon/height=WGS84/ellipsoidal,Q=1:fix,2:float,3:sbas,4:dgps,5:single,6:ppp,ns=# of satellites)
%  GPST                  latitude(deg) longitude(deg)  height(m)   Q  ns   sdn(m)   sde(m)   sdu(m)  sdne(m)  sdeu(m)  sdun(m) age(s)  ratio
2021/03/18 12:19:04.623   XX.848703631   XX.489500183   346.9086   1  16   0.0051   0.0037   0.0083   0.0012  -0.0016   0.0003  -0.00    5.8
2021/03/18 12:19:06.323   XX.848513143   XX.490020556   347.1328   1  16   0.0051   0.0037   0.0083   0.0012  -0.0016   0.0003  -0.00    5.7
```
2021/03/18 12:19:06.323 - next external event

rtklibexplorer, 2023-04-17:
Yes, RTKLIB can only support one external event per epoch.

closed 2023-04-17 by rtklibexplorer

a1ien, 2023-04-17:
It would be useful to be able to handle multiple events in the same epoch.