rtklibexplorer/RTKLIB #108: Fix status on console app (rtkrcv)

athtest800, 2022-01-28:
Hi,
When using windows binaries (rtknavi) or when compiling in ubuntu x86 and using rtkrcv, F9P gets FIX status in a minute.
In raspberry pi 4 I cannot get FIX status even if I export the .conf from ubuntu and importing to raspberry rtkrcv for similar configuration.
The test was also taken place simultaneously on ubuntu x86 and raspberry with the same feed but no FIX from raspberry whereas in ubuntu got FIX  in a minute?
Thank you

gaofenzi, 2022-01-28:
您好，辛苦啦！梦寒已经收到您的珍贵来件，会在最短的时间内给予您最想要的答复！

sonicpp, 2022-02-19:
I think I was facing the same issue when I was trying it on the raspberry - maybe it is too slow? Or is it because of different architecture? I would try to compile it without the "-g" (debugging symbols), it could speed it up little bit.

rtklibexplorer, 2023-04-17:
I have tested rtkrcv on multiple Raspberry Pi's without issue.  The faster Raspberry Pis are fast enough to run multiple instances of rtkrcv simultaneously.  I've posted more information on using rtkrcv on the Pi [here](https://rtklibexplorer.wordpress.com/2022/11/10/raspberry-pi-based-ppk-and-rtk-solutions-with-rtklib/).

closed 2023-04-17 by rtklibexplorer