rtklibexplorer/RTKLIB #110: Str2str error

ItsGettingWorse, 2022-03-10:
I'm having some issues with the CUI using str2str.
1. It seems that the forwarding speed is not enough. When forwarding the telegrams sent by other base stations, only the first telegram can be seen on the forwarded server, and other telegrams are not forwarded. And the accepted BPS shows a lower value.
2. There is no such problem in the win GUI.

gaofenzi, 2022-03-10:
您好，辛苦啦！梦寒已经收到您的珍贵来件，会在最短的时间内给予您最想要的答复！

ItsGettingWorse, 2022-03-17:
The problem has been found. The concept of CUI and GUI design is different. You can observe the "conv" array passed in by CUI and the "msg" string, and you will find the problem.
It is also very simple to modify. You can add the message number you need to the "msg" string according to the author's logic, or, like me, reconstruct the working logic of the "conv" array.

closed 2022-03-17 by ItsGettingWorse