rtklibexplorer/RTKLIB #115: Can a new GLONASS integer estimable AR method be implemented on rtklibexplorer?

KingGNSS, 2022-04-22:
Hello,
Recently, I read papers about an integer estimable AR method for GLONASS, but those papers describe estimating double-difference ambiguity directly, whereas rtklibexplorer actually estimates single-difference ambiguity from double-difference observations. I don't know if the new FMDA ambiguity integer estimability method mentioned in the following paper can be implemented in rtklibexplorer.

Those related papers:
1. Teunissen, P. J. G. "A new GLONASS FDMA model." GPS Solutions 23.4 (2019): 1-19.
2. Teunissen, P. J. G., and A. Khodabandeh. "GLONASS ambiguity resolution." GPS Solutions 23.4 (2019): 1-11.
3. Hou, Pengyu, Baocheng Zhang, and Teng Liu. "Integer-estimable GLONASS FDMA model as applied to Kalman-filter-based short-to long-baseline RTK positioning." GPS Solutions 24.4 (2020): 1-14.
4. Zaminpardaz, Safoora, Peter JG Teunissen, and Amir Khodabandeh. "GLONASS–only FDMA+ CDMA RTK: Performance and outlook." GPS Solutions 25.3 (2021): 1-12.
5. Brack, Andreas, Benjamin Männel, and Harald Schuh. "GLONASS FDMA data for RTK positioning: a five-system analysis." GPS solutions 25.1 (2021): 1-13.

rtklibexplorer, 2022-04-25:
Although the actual bias states in the RTKLIB kalman filter are single-differences, the filter updates and the ambiguity resolution are based on double differences.  That, along with the suggestion in the first paper in your list, that existing solutions can be fairly easily modified by multiplying the wavelength matrix by an additional term, suggest to me that it should be possible to apply this model to RTKLIB.  I would suggest to anyone interested in trying this, that my newly released python version of RTKLIB (rtklib-py) would be a good platform for experimenting with this idea.

KingGNSS, 2022-04-26:
> Although the actual bias states in the RTKLIB kalman filter are single-differences, the filter updates and the ambiguity resolution are based on double differences. That, along with the suggestion [...]

Thanks for your reply. Please leave me your email, I may need to discuss with you this GLONASS AR implementation in rtklibexplorer or rtklib-py, thank you.

KingGNSS, 2022-04-27:
> You can contact me at [rtklibexplorer@gmail.com](mailto:rtklibexplorer@gmail.com)

OK, I got it.

closed 2022-04-28 by KingGNSS