rtklibexplorer/RTKLIB #118: Virus Total Flags the demo5_b34f1.zip Release as Having the Gen:Variant.Lazy.194090 (or similar) by 9 Vendors

EdenRidgway, 2022-06-25:
I just checked the Mar 26, 2022 demo release (demo5_b34f1.zip) against VirusTotal and it flagged it as being "malicious" by 9 vendors:
https://www.virustotal.com/gui/file/60f92f945565cacfbbe0b39780ffb6b9af54715c02771f3297211a14217dddc8. While I suspect that these are false positives it is worth checking out.

I decided to go through each executable file in the zip and run them through VirusTotal. Here are the files with issues:
- **convbin.exe**: MaxSecure: Trojan.Malware.300983.susgen https://www.virustotal.com/gui/file/182fd2c5d57266ae51cfe2ba362875783658577f699d631df7678d9d800220ae
- **crx2rnx.exe**: SecureAge APEX: Malicious https://www.virustotal.com/gui/file/a5bf1806ba68fcea8f7c7c00609c08a35a3f1d13ff77066843cdbdc79e59b2f5
- **rtkpost.exe**: MaxSecure: Trojan.Malware.300983.susgen https://www.virustotal.com/gui/file/e87da661d3e25483837f628eeedf4a7fdaf459ace92e756c751bb53e721e1f31/detection
- **wget.exe**: SecureAge APEX: Malicious https://www.virustotal.com/gui/file/a963385f69c37fa6dc034d045aba1e7edf9bfb3ed6e4334a354cb822b3647340

These files had no issues:
- **pos2kml.exe**: https://www.virustotal.com/gui/file/37fca92e1480987a5dcfaac154a241f58f577061cf84327c9303af92837015ea
- **rnx2rtkp.exe**: https://www.virustotal.com/gui/file/82247d5d8a14de1f71168f660dbbd8faf67b33468474d886fc8443c301f50d2e
- **rtkconv.exe**: https://www.virustotal.com/gui/file/14fe3bdb7d24528c6f4da953b45d01021f6aeddf41002dc26fc3ef97838c30e7
- **rtkget.exe**: https://www.virustotal.com/gui/file/924873a7f7ea2baf5e9389f298b95b576d02bfe1624343aadce58db6df2c7486
- **rtklaunch.exe**: https://www.virustotal.com/gui/file/408c28e7d59d4a3c0192996fbfc2689248f502d61e84d38f9ce152f5c5a05af8
- **rtknavi.exe**: https://www.virustotal.com/gui/file/6dc16bd21503378d6cc835d10ff6f6af83a633961579cd879190fde1face838c
- **rtkplot.exe**: https://www.virustotal.com/gui/file/02bcebf28621c440f70593da33880d2d356e71c7c42bde8ca2ea871665484e4c
- **srctblbrows.exe**: https://www.virustotal.com/gui/file/08492881231bc9f6255cc6632cb0ffd4990c92d7024e7f569d3c456679051666
- **strsvr.exe**: https://www.virustotal.com/gui/file/b1702087e3ea50bb836de034662a71341d7ecf8082057f3565b74086b7a66fb9

It is interesting that the scan of the zip file throws up more issues than just scanning the individual executables. Furthermore the "virus" detected is different when scanning the executables and the zip file.

gaofenzi, 2022-06-25:
您好，辛苦啦！梦寒已经收到您的珍贵来件，会在最短的时间内给予您最想要的答复！

rtklibexplorer, 2023-04-17:
I am fairly certain these are all false positives.  I have made some changes to try and reduce these.  I just ran the latest code (b34g) through VirusTotal and it was deemed safe by all 89 scanners.

closed 2023-04-17 by rtklibexplorer