rtklibexplorer/RTKLIB #123: qt?

paapu88, 2022-08-01:
Dear Developers,

For graphical interface one seems to need qt.
Any instructions for qt? should it be version 5.15 ?

Terveisin Markus

ignabelitzky, 2022-08-02:
I would start with Qt6... version 6.3.1 works fine to me and have several changes in comparison to 5.15

I would start doing some projects, there are several books out there... If you just started I recommend you Foundations of Qt Development by Johan Thelin... also join the community in Discord... they are grate people!!

paapu88, 2022-08-03:
Thanks,
Compiled qt 6.3.1, it seems to be fine.
But when doing
I get after 
1) cd app/qtapp
2) qmake
3) make
it compiles some time but crashes then with:
```
getmain.cpp: In member function ‘void MainForm::BtnTime1Click()’:
getmain.cpp:453:40: error: ‘class QDateTime’ has no member named ‘toTime_t’
  453 |     t1.time = static_cast<time_t>(time.toTime_t()); t1.sec = time.time().msec() / 1000;

```

ignabelitzky, 2022-08-04:
I think you should replace "time.toTime_t()" with "time.toSecsSinceEpoch()" to work properly... [here](https://doc.qt.io/qt-5/qdatetime-obsolete.html) explains that toTime_t() is obsolete for newer versions... hope it helps you

paapu88, 2022-08-15:
Thanks, I dont want to debug other peoples code, so I'm using wine to run windows binary of rtklib 2.4.3 from https://github.com/tomojitakasu/RTKLIB_bin/tree/rtklib_2.4.3

closed 2022-08-15 by paapu88