rtklibexplorer/RTKLIB #125: MacOS installation

rvizcarra-eclipse, 2022-08-22:
Attempted to install on OS X but got this error

ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [../lib/libRTKLib.1.0.0.dylib] Error 1
make: *** [sub-------src-make_first] Error 2

charliegilliland, 2023-12-09:
On MacOS librt isn't available, so you can edit each makefile and remove ```-lrt``` from LDLIBS. I'm compiling console apps successfully with this edit.

adamgarbo, 2024-02-18:
I can confirm I'm also able to successfully compile on macOS (12.7.1) by removing `-lrt` from the makefile. @reinevizcarra I reckon this should resolve this issue.

HamishB, 2025-03-13:
FYI I have successfully compiled the convbin, pos2kml, and rnx2rtkp command line (consapp) programs on MacOS 15 (ARM) after installing XCode, XCode command line tools, and gfortran (fxcoudert's github installer).

rtkrcv and str2str complain about undeclared baud rate definitions and `hp->h_addr` in stream.c, but I don't need those so I didn't try for a fix.

As MacOS's cc uses clang not glibc libraries I had to manually declare the snprintf() function prototype in src/options.c's str2enum() as:
`int snprintf(char *buf, size_t size, const char *fmt, ...);`

and edit the app/consapp/rnx2rtkp/gcc/makefile to tell it where to find the gfortran library:
`LDLIBS = -L/usr/local/gfortran/lib`

have fun,
Hamish