rtklibexplorer/RTKLIB #135: Unable to compile all qt apps in Linux

escuta, 2023-01-21:
Hi, I know there's a warning the readme that the qt apps may not work, however, in the demo5 branch, I've been unable to compile rtkpost_qt, rtknavi_qt, rtkconv_qt and rtkpost_qt. All command line apps compile.

The errors include:

```
postmain.cpp: In member function ‘int MainForm::GetOption(prcopt_t&, solopt_t&, filopt_t&)’:
postmain.cpp:948:12: error: ‘struct prcopt_t’ has no member named ‘maxgdop’
  948 |     prcopt.maxgdop  =RejectGdop;
      |            ^~~~~~~
postmain.cpp:949:21: error: incompatible types in assignment of ‘double’ to ‘double [2]’
  949 |     prcopt.maxinno  =RejectThres;
```
and 

```
rcv/septentrio.c: In function ‘decode_measextra’:
rcv/septentrio.c:302:36: warning: unused parameter ‘raw’ [-Wunused-parameter]
  302 | static int decode_measextra(raw_t *raw)
      |                             ~~~~~~~^~~
make[1]: *** [Makefile:894: postmain.o] Error 1
make[1]: Saindo do diretório '/home/iain/git/RTKLIB/app/qtapp/rtkpost_qt'

make: *** [Makefile:155: sub-rtkpost_qt-make_first] Error 2
```
and 

```
navimain.cpp: In member function ‘void MainWindow::LoadOpt()’:
navimain.cpp:2425:12: error: ‘struct prcopt_t’ has no member named ‘maxgdop’
 2425 |     PrcOpt.maxgdop = settings.value("prcopt/maxgdop", 30.0).toDouble();
      |            ^~~~~~~
navimain.cpp:2426:20: error: incompatible types in assignment of ‘double’ to ‘double [2]’
 2426 |     PrcOpt.maxinno = settings.value("prcopt/maxinno", 30.0).toDouble();
      |     ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
navimain.cpp: In member function ‘void MainWindow::SaveOpt()’:
navimain.cpp:2639:48: error: ‘struct prcopt_t’ has no member named ‘maxgdop’
 2639 |     settings.setValue("prcopt/maxgdop", PrcOpt.maxgdop);
      |                                                ^~~~~~~
navimain.cpp:2640:22: error: use of deleted function ‘QVariant::QVariant(void*)’
 2640 |     settings.setValue("prcopt/maxinno", PrcOpt.maxinno);
      |     ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/qt/QtCore/qsettings.h:44,
                 from /usr/include/qt/QtCore/QSettings:1,
                 from navimain.cpp:38:
/usr/include/qt/QtCore/qvariant.h:499:12: note: declared here
  499 |     inline QVariant(void *) = delete;
      |            ^~~~~~~~
      
```
and

```
naviopt.cpp: In member function ‘void OptDialog::GetOpt()’:
naviopt.cpp:472:33: error: ‘struct prcopt_t’ has no member named ‘maxgdop’
  472 |     RejectGdop->setValue(PrcOpt.maxgdop);
      |                                 ^~~~~~~
naviopt.cpp:473:34: error: cannot convert ‘double [2]’ to ‘double’
  473 |     RejectThres->setValue(PrcOpt.maxinno);
      |                           ~~~~~~~^~~~~~~
      |                                  |
      |                                  double [2]
In file included from /usr/include/qt/QtWidgets/QDoubleSpinBox:1,
                 from ui_naviopt.h:17,
                 from naviopt.h:6,
                 from naviopt.cpp:12:
/usr/include/qt/QtWidgets/qspinbox.h:171:26: note:   initializing argument 1 of ‘void QDoubleSpinBox::setValue(double)’
  171 |     void setValue(double val);
      |                   ~~~~~~~^~~
naviopt.cpp: In member function ‘void OptDialog::SetOpt()’:
naviopt.cpp:600:12: error: ‘struct prcopt_t’ has no member named ‘maxgdop’
  600 |     PrcOpt.maxgdop = RejectGdop->value();
      |            ^~~~~~~
naviopt.cpp:601:20: error: incompatible types in assignment of ‘double’ to ‘double [2]’
  601 |     PrcOpt.maxinno = RejectThres->value();
      |     ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
naviopt.cpp: In member function ‘void OptDialog::LoadOpt(const QString&)’:
naviopt.cpp:785:33: error: ‘struct prcopt_t’ has no member named ‘maxgdop’
  785 |     RejectGdop->setValue(prcopt.maxgdop);
      |                                 ^~~~~~~
naviopt.cpp:786:34: error: cannot convert ‘double [2]’ to ‘double’
  786 |     RejectThres->setValue(prcopt.maxinno);
      |                           ~~~~~~~^~~~~~~
      |                                  |
      |                                  double [2]
/usr/include/qt/QtWidgets/qspinbox.h:171:26: note:   initializing argument 1 of ‘void QDoubleSpinBox::setValue(double)’
  171 |     void setValue(double val);
      |                   ~~~~~~~^~~
naviopt.cpp: In member function ‘void OptDialog::SaveOpt(const QString&)’:
naviopt.cpp:1000:12: error: ‘struct prcopt_t’ has no member named ‘maxgdop’
 1000 |     prcopt.maxgdop = RejectGdop->value();
      |            ^~~~~~~
naviopt.cpp:1001:20: error: incompatible types in assignment of ‘double’ to ‘double [2]’
 1001 |     prcopt.maxinno = RejectThres->value();
      |     ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
    
 ```
Thanks

gaofenzi, 2023-01-21:
您好，辛苦啦！梦寒已经收到您的珍贵来件，会在最短的时间内给予您最想要的答复！

escuta, 2023-01-21:
I see below that HamishB was right. If I go back to the commit before the one he mentions, all qt apps compile, a relief!

https://github.com/rtklibexplorer/RTKLIB/issues/127#issuecomment-1296463732

rtklibexplorer, 2023-04-17:
Compile errors fixed with commit on 4/17/23.  Note that the qt apps are still not supported but at least should compile now.  The user interfaces do not support many of the parameters used by the new features in the demo5 code

closed 2023-04-17 by rtklibexplorer