rtklibexplorer/RTKLIB #136: Bug in rnx2rtkp, -h flag does not work

pepijntje02, 2023-02-21:
Hi, 

I tried to process my RINEX files wit different methods. This is done to figure out which combination is most reliable for accurate measurements. 

First all data was processed using the GUI, but I got tiered of clicking buttons so I switched to the command line. While processing the data I noticed that there was no difference between the ambiguity solution using `continuous` or `fix and hold`. 

The commands I used for continuous and fix_and_hold (in that order):
`rnx2rtkp.exe -r 3899613.9605, 397361.6236, 5014738.6355 -ts 2023/02/06 17:32 -te 2023/02/06 17:38 -p 2 -f 3 -c -sys 13 rover.23o reference.23o rover.23n -o continuous.pos`
`rnx2rtkp.exe" -r 3899613.9605, 397361.6236, 5014738.6355 -ts 2023/02/06 17:32 -te 2023/02/06 17:38 -p 2 -f 3 -c -sys 13 -h rover.23o reference.23o rover.23n -o fix_and_hold.pos`

The output files are exactly the same. 

When I add `-i` flag for instantaneous solutions there were not RTK fixes, all float (Q=2). 

How can I check if `continuous` is used, or if `fix_and_hold` is used? The text in the `.pos` file did change (see `amb res`):

```
% pos mode  : Kinematic
% freqs     : L1+L2/E5b+L5
% solution  : Combined-Phase Reset
% elev mask : 10.0 deg
% dynamics  : on
% tidecorr  : off
% ionos opt : Broadcast
% tropo opt : Saastamoinen
% ephemeris : Broadcast
% navi sys  : GPS GLONASS Galileo
% amb res   : Continuous
% amb glo   : Fix and Hold
% val thres : 3.0
```
and
```
% freqs     : L1+L2/E5b+L5
% solution  : Combined-Phase Reset
% elev mask : 10.0 deg
% dynamics  : on
% tidecorr  : off
% ionos opt : Broadcast
% tropo opt : Saastamoinen
% ephemeris : Broadcast
% navi sys  : GPS GLONASS Galileo
% amb res   : Fix and Hold
% amb glo   : Fix and Hold
% val thres : 3.0
```
:

gaofenzi, 2023-02-21:
您好，辛苦啦！梦寒已经收到您的珍贵来件，会在最短的时间内给予您最想要的答复！

rtklibexplorer, 2023-04-17:
Your header output does confirm that "Fix and Hold" was selected.  A hold will only occur after a specified number of consecutive fixes and a minimum number of satellites for the hold epoch.  Most likely your solution never met these criteria, so a hold never occurred, in which case your output would be identical to a continuous solution.   For debug purposes, you can lower the AR threshold and the minimum hold sats threshold to confirm that you see a difference.

closed 2023-04-17 by rtklibexplorer