rtklibexplorer/RTKLIB #140: `solopt[1]` options not initialized to default in `rtkrcv`

jacqu, 2023-04-18:
Hi!

`solopt[1]` options are not initialized to the default value when running `rtkrcv`. This yields for example a timestamp printed with insufficient resolution for solutions sent through a socket:
```
2258 143646   48.526389460    7.736927026   214.7750   1  12   0.0270   0.0193   0.0440  -0.0115  -0.0122  -0.0107   0.79    3.9
2258 143646   48.526389446    7.736926981   214.7699   1  12   0.0270   0.0193   0.0440  -0.0115  -0.0122  -0.0107   0.99    3.9
2258 143646   48.526389489    7.736926957   214.7809   1  12   0.0270   0.0193   0.0440  -0.0115  -0.0122  -0.0107   1.19    3.9
2258 143646   48.526389441    7.736927036   214.7698   1  12   0.0270   0.0192   0.0439  -0.0115  -0.0122  -0.0107   0.39    3.9
2258 143647   48.526389422    7.736927035   214.7771   1  12   0.0270   0.0193   0.0440  -0.0115  -0.0122  -0.0107   0.59    3.9
2258 143647   48.526389414    7.736927020   214.7749   1  12   0.0270   0.0193   0.0440  -0.0115  -0.0122  -0.0107   0.79    3.9
2258 143647   48.526389433    7.736927005   214.7653   1  12   0.0270   0.0193   0.0440  -0.0115  -0.0122  -0.0107   0.99    3.9
2258 143647   48.526389419    7.736927010   214.7722   1  12   0.0270   0.0193   0.0440  -0.0115  -0.0122  -0.0107   1.19    3.8
2258 143647   48.526389439    7.736927033   214.7752   1  12   0.0271   0.0193   0.0441  -0.0116  -0.0122  -0.0107   1.39    3.8
2258 143648   48.526389431    7.736927013   214.7666   1  12   0.0270   0.0192   0.0440  -0.0115  -0.0122  -0.0107   0.59    3.8
2258 143648   48.526389398    7.736927061   214.7718   1  12   0.0270   0.0193   0.0440  -0.0115  -0.0122  -0.0107   0.79    3.8
2258 143648   48.526389445    7.736927058   214.7700   1  12   0.0270   0.0193   0.0440  -0.0115  -0.0122  -0.0107   0.99    3.8
```

When forcing `solopt[1]` to point to the same location than `solopt[0]`, it fixes the timestamp resolution issue since the default `timeu` value is 3:
```
2023/04/18 22:17:56.000   48 53 03.08597    7 33 02.82913   197.9141   2  10   0.2733   0.2019   0.9704  -0.0537   0.0814  -0.3483   1.00    1.5
2023/04/18 22:17:56.200   48 53 03.08577    7 33 02.82907   197.9426   2  10   0.2714   0.2006   0.9635  -0.0534   0.0810  -0.3458   1.20    1.5
2023/04/18 22:17:56.400   48 53 03.08595    7 33 02.82915   197.9393   2  10   0.2696   0.1993   0.9566  -0.0531   0.0806  -0.3434   0.40    1.5
2023/04/18 22:17:56.600   48 53 03.08593    7 33 02.82889   197.9692   2  10   0.2678   0.1980   0.9498  -0.0528   0.0803  -0.3409   0.60    1.5
2023/04/18 22:17:56.800   48 53 03.08585    7 33 02.82872   197.9850   2  10   0.2661   0.1967   0.9432  -0.0525   0.0799  -0.3386   0.80    1.5
2023/04/18 22:17:57.000   48 53 03.08578    7 33 02.82907   198.0185   2  10   0.2643   0.1955   0.9367  -0.0523   0.0795  -0.3363   1.00    1.6
2023/04/18 22:17:57.200   48 53 03.08579    7 33 02.82894   198.0350   2  10   0.2626   0.1943   0.9303  -0.0520   0.0792  -0.3340   1.20    1.6
2023/04/18 22:17:57.400   48 53 03.08592    7 33 02.82913   198.0549   2  10   0.2610   0.1931   0.9241  -0.0517   0.0788  -0.3317   1.40    1.6
```

I'm not sure that my proposed patch below has no side effects; My understanding of the option mechanism is still incomplete. So consider this as a draft proposition that should be carefully reviewed:

In `rtkrcv.c`, add the following line after line `1660 getsysopts(&prcopt,solopt,&filopt); `
`getsysopts(&prcopt,solopt+1,&filopt);`

This will make `solopt[0] == solopt[1]`
Valgrind seems to be happy with this:
```
rtk server shutdown ...
==382356== 
==382356== HEAP SUMMARY:
==382356==     in use at exit: 9,289,344 bytes in 391 blocks
==382356==   total heap usage: 6,725 allocs, 6,334 frees, 93,093,629 bytes allocated
==382356== 
==382356== LEAK SUMMARY:
==382356==    definitely lost: 0 bytes in 0 blocks
==382356==    indirectly lost: 0 bytes in 0 blocks
==382356==      possibly lost: 0 bytes in 0 blocks
==382356==    still reachable: 9,289,344 bytes in 391 blocks
==382356==         suppressed: 0 bytes in 0 blocks
==382356== Rerun with --leak-check=full to see details of leaked memory
==382356== 
==382356== For lists of detected and suppressed errors, rerun with: -s
==382356== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

Best,
Jacques

jacqu, 2023-04-19:
I think I should post this on the main git repo...

garrettseepersad, 2023-04-28:
Hello @jacqu , do you know any good tutorials or could you share the valgrind commands for assessing RTKlib? I have zero experience with valgrind. It's been on my "things to learn" list

jacqu, 2023-04-28:
Hi Garret,

Valgring is a runtime memory access checker. You just need to compile your code with the -g option, type « valgring name_of_the_app_that_needs_to_be_checked » and look for messages indicating faulty memory access. 

Best,
Jacques

rtklibexplorer, 2024-08-07:
I believe this is resolved by PR #384

closed 2024-08-07 by rtklibexplorer