rtklibexplorer/RTKLIB #141: Integer Ambiguity Resolution for Galileo (and more)

epigramx, 2023-04-29:
I noticed that there's a plethora of Galileo satellites currently in easy view from many locations (e.g. according to https://in-the-sky.org/satmap_worldmap.php?year=2023&month=4&day=29), and I was wondering if that can be a feature somehow (if there's a mathematical limitation preventing it let me know).

Also I was thinking that the current system appears to mainly work with GPS; is there a technological limitation that prevents e.g. having GLONASS as the "primary" constellation for integer ambiguity solving or even Galileo? Also for some reason I might get better fixes if fix-and-hold on glonass is not used.

epigramx, 2023-05-01:
I might have discovered a but in your code related to this. GLONASS integer ambiguity appears to never be turned off at `ddidx()` of `rtkpost.c`. If I turn it off manually at  the top of the loop of that function it appears to produce results not found with the gui option set to off.

epigramx, 2023-05-04:
For the time being I have concluded that the code just does integer ambiguity by default for the major constellations anyway (at least if no configuration option denies that in `ddidx()` of `rtkpost.c`), so I don't see a reason to update anything in terms of basic constellation support for it (unless it's for avoiding bugs for certain constellations or tweaking their default related constants). Let me know if I'm wrong (and I'm possibly wrong).

rtklibexplorer, 2024-03-16:
I believe your last summary is basically correct, the AR_MODE switch turns all ambiguity resolution on or off and the other switches (GLO_AR_MODE and BDS_AR_MODE) can be used to disable specific constellations.

closed 2024-03-16 by rtklibexplorer

epigramx, 2024-04-26:
> the other switches (GLO_AR_MODE and BDS_AR_MODE) can be used to disable specific constellations.

@rtklibexplorer at line: https://github.com/rtklibexplorer/RTKLIB/blob/2738ae84d0e4e05a2b4ea0d3ad72331711db9aa8/src/rtkpos.c#L1467
it appears that while `opt.bdsmodear` is respected to do a relevant skipping: I don't see `GLO_ARMODE_OFF` being respected there,

so I was testing with a couple of lines of:

```
        if (m == 1 && rtk->opt.glomodear == GLO_ARMODE_OFF) continue;
        if (m == 3 && rtk->opt.bdsmodear == 0) continue;
```
above the nofix (linked) line and it appears to produce different/better results in some cases which is why I had originally posted above "If I turn it off manually at the top of the loop of that function it appears to produce results not found with the gui option set to off" and "I don't see a reason to update anything in terms of basic constellation support for it **(unless it's for avoiding bugs for certain constellations or tweaking their default related constants)**".