rtklibexplorer/RTKLIB #142: SSL/TLS support

thesourcerer8, 2023-06-08:
The european Galileo HAS (High-Accuracy-Service) is providing a SSL/TLS enabled NTRIP caster on port 443, therefore I would like to have SSL/TLS support added to RTKLIB. I would be willing to develop a pull-request for this, but would like to know about the chances to get it accepted before I start the work.

rtklibexplorer, 2023-06-08:
Hi there,

That would be great!  I'd be happy to pull the code.

Tim

DavidKelleySCSC, 2023-06-08:
Tim and thesourcerer8

The timing on this is really perfect.  I would love to assist with this in any way that helps.  That 'best practices' advice document from RTCM on how to build a good NTRIP Client should be finally published any day now and has some basic SSL/TLS advice in it.  A key take away is if you use OpenSSL all you really have to implement is a basic way for the NTRIP Client user to look at the certificates and accept any exceptions on a _caster-by-caster_ basis when there is a need. 

I am expecting a major push to get folks to use NTRIP Rev2 (and SSL/TLS enabled connection) for the rest of the year.  And I have really wanted to get all the RTKLIB code deployment to be part of that. And as rtklibexplorer had already created code to overcome the "do not connect when you have nothing to say" problem, and has progressive back off time (two major problems with RTKLIB), I can think of no better place to start than here.

thesourcerer8, 2023-06-10:
Ok, great! I did an initial investigation of the RTKLIB codebase, and found 2 references to libssl:
pp/qtapp/strsvr_qt/out:--32245-- Reading syms from /usr/lib64/libssl.so.1.0.2g
app/qtapp/rtkget_qt/out:--18097-- Reading syms from /usr/lib64/libssl.so.1.0.2g
but that seems to be pulled in by the QT library only, so I guess the challenge will be to reuse the libssl library provided with QT on the QT platforms and to investigate the best options on all the other platforms.

DavidKelleySCSC, 2024-10-20:
I would like to get this advanced and think the GUI element is simply a 'use a secure connection" button next to the "use Version 2" button in the GUI.  And the Version 2 support still needs to be added before this can be completed. [And that button should only be active if NTRIP Version 2 is selected as there is not concept of a secure connection in Version 1] 

I think that regardless of using Qt of just C/C++ you will end up using the openSSl library (aka libssl.so.* or libssl-1_1*.dll for windows) as just about everyone seems to implement SSL/TLS using that for the heavy lifting.  This includes all the major GNSS makers in their NTRIP clients (Trimble, TopCon, Septentrio, etc.).

I have no practical experience using it with C/C++ projects (we use Qt to develop SNIP and other NTRIP work).  In Qt the last few 'free' Editions (Qt 5) it uses a somewhat outdated edition of openSSL, while the "you must pay us or else" release (Qt 6.x) uses the latest.

Most security certificates have four dozen or more cypher suites to chose from so I do not think that will be an issue any time soon.  [Aside, if you want to test a secure socket NTRIP connection use rtk2go.com on port **2102**  (not 2101) and you will get a working, but self-signed, cert as part of the connection.  We rotate the CERT every month or so so people can test things on that host.

In terms of building this out for the Qt edition of RTKLIB, I can provide a few support dialogs if there is interest. [You need to pop up an 'are you sure' dialog if the returned security certificate is not fully valid or does not come for a known trusted source, and then you need to display the details about it for the user to review]  Not sure how doing that would work for non-Qt builds which are still the majority of the users.

DavidKelleySCSC, 2024-10-24:
I downloaded the latest with the Qt part in it, but found that would only build for a Linux machine (and presumably using the older Qt5.x which is fine).  But we have evolved into a Windows shop anymore (please save the pity comments) so a Windows copy of RTKLIB-Qt would be best from my own needs so I do not have to  root thorough a bunch of re-defined handle pointer issues before any work can be done.  **Does such a thing exist?**  

Looking at the repository that Jens Reimann ran for RTKLIB-Qt it is now marked obsolete and have not been updated for several years.  I do not see anything more current to start from.

timsurber, 2024-10-24:
What do you guys think about moving the whole NTRIP code to libcurl instead of the manual TCP requests? We would get SSL support "for free"

JensReimann, 2025-01-03:
Dear DavidKelleySCSC,
I just read your comments to this ticket. The Qt GUI is now developed in this repository and has heavily updated compared to my own repository.

The GUI should compile with Qt5 and Qt6 on Windows and Linux. Unfortunately, I cannot test every configuration. SO please file a bug report, if there is any issues.

ZhangRunzhi20, 2025-03-11:
OpenSSL provides excellent support for SSL/TLS. However, configuring the environment on Windows can be quite complex, as users need to install and set it up manually. That said, one of its key advantages is the convenient support for TLS.

We have developed an NTRIP client with TLS support based on stream.c from RTKLIB, specifically for Linux systems (https://github.com/ZhangRunzhi20/RTStreamHub). We welcome contributions and discussions on the best approach to implementing NTRIP TLS support in RTKLIB.