rtklibexplorer/RTKLIB #165: BDS Data Reading in RTCM3 and RINEX 3.04 Data Decode issue

sean10776, 2023-12-15:
I found some issue at rinex 3.04 decoding and rtcm3 BDS decoding.

### Issues
1. RTCM3 BDS frquency index is set `B6*` to `2` and `B5*` to `3`, but the `codepris` bds freq 2 corresponds to B5's code type, and freq 3 corresponds to B3's code type.
2. RINEX3.04 decoding already set the freq index for each split value at `set_index`, but `decode_obsdata` set the freq index again. Additionally, assign postion in the` decode_obsdata` only assigns the pseudorange index, neglecting other indices such as carrier phase and snr. I found this issue by B2I and B1P observe by the same satellite at time.

### My Current Approach
1. Reset freq index `B6*` to `3` and `B5*` to `2`
2. Comment out the assign position part in  `decode_obsdata`

English is not my native language. If there are any misunderstandings or language errors, please feel free to inform me.

rtklibexplorer, 2023-12-15:
Thanks for pointing these out.  I understand issue #1. but am struggling to completely understand #2.  If you were able to send me the exact code change you made and a sample observation file that demonstrates the problem, that would be very helpful.  You can send it to me at rtklibexplorer@gmail.com

sean10776, 2023-12-16:
Copy that. I just sent the email.

rtklibexplorer, 2024-01-02:
I've just commited a fix based on your code suggestions.  Re item #1, I've switched the priorities to match the new freq indices rather than restoring the previous freq indices

closed 2024-01-06 by rtklibexplorer