rtklibexplorer/RTKLIB #168: time-interpolation of residuals skipping after half the time span, was this intended?

ourairquality, 2024-01-18:
Noticed the carrier phase residuals jumping periodically, on a ppk solution with the rover sampling ever 1 second and the base every 30 seconds there were jumps in the middle of every 30 second period. Tracked it down to the time-interpolation of residuals, the function rtkpos.c:intpres(), and to the code at the end of this function:

```
    if (fabs(tt)<fabs(ttb)) {
        nb=n;for (i=0;i<n;i++) obsb[i]=obs[i];  /* copy obs to previous base obs */
        return tt;
    } else return ttb;

```

This has the effect of jumping to the current base value after half of the period , rather than continuing the linear interpolation between the current and previous value.

This was introduced in commit 4b519f36, and reverting to the previous ending cleaned up these periodic jumps:

`    return fabs(ttb)>fabs(tt)?ttb:tt;
`

Just querying this in case it was unintended.

rtklibexplorer, 2024-02-22:
This was part of a change to use the closest base observation rather than the closest previous base observation for ppk solutions.  In my test case the current code interpolates base observations between -15 sec and +15 sec and with your suggested change it reverts to interpolating between -30 sec and 0 sec. In my case, the current code gives better results and I don't see the residual jumps you mention.  However, it's very possible your data is exercising the code in a different way.  If possible, can you send an example data set with config file to rtklibexplorer@gmail.com and I will take a closer look.

ourairquality, 2024-02-25:
The case that showed the issue was with the rover sampling at 1 sec and the base at 30 sec.

The interpolation is between two base time sets of observation residuals, and these are separated by 30 sec in this case. With the change reverted, looking at a cycle of 30 sec as the rover time increases,  'tt' varies -29 -28 ... -1 sec, 'ttb' varies 1 2 ... 29 sec, and the returned delta varies -29 -28 ... -15 16 17 ... 29 respectively. The rover time is between the two sets of base observations and 'tt' and 'ttb' are used to linearly weight the residuals. This interpolation seem okay.

The returned delta appears to only be passed to varerr() as a variance input. It is squared before adding to the variance so could be negative or positive. Perhaps returning the maximum absolute time delta here was not so useful. It seems intuitive that as the rover time approaches the base observation times that the error would be less that when in between, whereas in the prior code this is the opposite of the return delta weight.

The current code sets the obsb set to the obs set at the half way point. Both sets are the some after this, so there is no interpolation, it is the same as using the current base observations after half way. It is interpolation for the first half of the period and then a jump to the future base set. One way to see this is tracing 'ttb', and see that it equals 'tt' in the second half, because the two sets are the same, 'ttb' varies 1 2 ... 16 -13 -12 ... -1.

The current code returns a delta of 1 2 ... 15 -14 -13 ... -1 respectively. If this were the intended change then returning the min absolute value of 'tt' and 'ttb' would fix that without breaking the interpolation. E.g. `return fabs(ttb)<fabs(tt)?ttb:tt`.

If an example of the steps in the residuals is needed then I can look into that.

ourairquality, 2024-02-25:
Here is an example of the jumps in the residuals with the current code.
[image]
[image]

closed 2024-02-26 by ourairquality