rtklibexplorer/RTKLIB #186: Duplicate in .sbf to rinex conversion

Stefal, 2024-02-11:
Hi!

If I try convert a binary file from a Septentrio Mosaic X5 to a rinex file, there are a lot of duplicates epoch in the file.
Here is a simplified example : 

```
> 2024 02 10 00 06 00.0000000  0 10                     
G19  21340104.155    21340102.818  
G28  24437995.427    24437995.204  
G31  23565093.391    23565092.998  
G06  23138826.912    23138826.716  
G17  21584440.797    21584440.373  
G21  24586926.184    24586924.986  
G04  20705745.798    20705745.477  
G02  23137518.998    23137517.948  
G03  20720695.771    20720695.561  
G09  22666964.109    22666963.609  
> 2024 02 10 00 06 30.0000000  0 10                     
G19  21338383.484    21338382.139  
G28  24445068.545    24445068.295  
G31  23563260.367    23563259.987  
G06  23122118.655    23122118.459  
G17  21592845.165    21592844.741  
G21  24604949.881    24604948.679  
G04  20699374.973    20699374.643  
G02  23154332.892    23154331.833  
G03  20730714.503    20730714.298  
G09  22648742.218    22648741.727  
> 2024 02 10 00 06 30.0000000  0 10                     
G19  21338383.484    21338382.139  
G28  24445068.545    24445068.295  
G31  23563260.367    23563259.987  
G06  23122118.655    23122118.459  
G17  21592845.165    21592844.741  
G21  24604949.881    24604948.679  
G04  20699374.973    20699374.643  
G02  23154332.892    23154331.833  
G03  20730714.503    20730714.298  
G09  22648742.218    22648741.727  
> 2024 02 10 00 06 30.0000000  0 10                     
...
> 2024 02 10 00 06 30.0000000  0 10                     
...
> 2024 02 10 00 06 30.0000000  0 10                     
...
> 2024 02 10 00 06 30.0000000  0 10                     
...
> 2024 02 10 00 06 30.0000000  0 10                     
...
> 2024 02 10 00 06 30.0000000  0 10                     
...
> 2024 02 10 00 06 30.0000000  0 10                     
...
> 2024 02 10 00 06 30.0000000  0 10                     
...
> 2024 02 10 00 06 30.0000000  0 10                     
...
> 2024 02 10 00 06 30.0000000  0 10                     
...
> 2024 02 10 00 07 00.0000000  0 10                     
G19  21336794.252    21336792.911  
G28  24452249.487    24452249.219  
G31  23561544.860    23561544.480  
G06  23105440.874    23105440.669  
G17  21601351.916    21601351.501  
G21  24623000.022    24622998.811  
G04  20693111.712    20693111.386  
G02  23171197.743    23171196.684  
G03  20740795.910    20740795.705  
G09  22630575.312    22630574.816  
```
There are usually 11 to 13 duplicates.
Same behaviour with a 1s interval:
```
> 2024 02 10 18 38 50.0000000  0 11
> 2024 02 10 18 38 51.0000000  0 11
> 2024 02 10 18 38 52.0000000  0 11
> 2024 02 10 18 38 53.0000000  0 11
> 2024 02 10 18 38 54.0000000  0 11
> 2024 02 10 18 38 54.0000000  0 11
> 2024 02 10 18 38 54.0000000  0 11
> 2024 02 10 18 38 54.0000000  0 11
> 2024 02 10 18 38 54.0000000  0 11
> 2024 02 10 18 38 54.0000000  0 11
> 2024 02 10 18 38 54.0000000  0 11
> 2024 02 10 18 38 54.0000000  0 11
> 2024 02 10 18 38 54.0000000  0 11
> 2024 02 10 18 38 54.0000000  0 11
> 2024 02 10 18 38 54.0000000  0 11
> 2024 02 10 18 38 54.0000000  0 11
> 2024 02 10 18 38 55.0000000  0 11
> 2024 02 10 18 38 56.0000000  0 11
> 2024 02 10 18 38 57.0000000  0 11
> 2024 02 10 18 38 58.0000000  0 11
> 2024 02 10 18 38 59.0000000  0 11
> 2024 02 10 18 39 00.0000000  0 11
> 2024 02 10 18 39 00.0000000  0 11
> 2024 02 10 18 39 00.0000000  0 11
> 2024 02 10 18 39 00.0000000  0 11
> 2024 02 10 18 39 00.0000000  0 11
> 2024 02 10 18 39 00.0000000  0 11
> 2024 02 10 18 39 00.0000000  0 11
> 2024 02 10 18 39 00.0000000  0 11
> 2024 02 10 18 39 00.0000000  0 11
> 2024 02 10 18 39 00.0000000  0 11
> 2024 02 10 18 39 00.0000000  0 11
> 2024 02 10 18 39 00.0000000  0 11
> 2024 02 10 18 39 01.0000000  0 11
> 2024 02 10 18 39 02.0000000  0 11
> 2024 02 10 18 39 03.0000000  0 11
```

I can send you a sbf file if you need it, or a realtime stream.

rtklibexplorer, 2024-02-11:
Yes, please send an sbf file to rtklibexplorer@gmail.com and I will take a look.  Thanks.

rtklibexplorer, 2024-02-13:
Changed default time tolerance from 0.0 to 0.005 in CONVBIN to avoid the duplicate epochs and to match the default value in RTKCONV

closed 2024-02-13 by rtklibexplorer

Ehabh, 2026-08-21:
Just want to add that I ran into what appears to be the same underlying issue, but with u‑blox RXM‑RAWX data rather than SBF. 
In the official tomojitakasu/RTKLIB master branch, roughly 40–50% of satellites were duplicated within single epochs across many independent hourly logs. After switching to the demo5 branch (tested at commit 75a2e56), the problem disappeared entirely, that is the exact same input files converted cleanly with no duplication.