rtklibexplorer/RTKLIB #209: rtkplot_qt build error

Bzz2745, 2024-03-25:
生成开始于 14:56...
1>------ 已启动生成: 项目: rtkconv_qt, 配置: Debug x64 ------
2>------ 已启动生成: 项目: strsvr_qt, 配置: Debug x64 ------
3>------ 已启动生成: 项目: srctblbrows_qt, 配置: Debug x64 ------
4>------ 已启动生成: 项目: rtkplot_qt, 配置: Debug x64 ------
5>------ 已启动生成: 项目: rtkget_qt, 配置: Debug x64 ------
6>------ 已启动生成: 项目: rtknavi_qt, 配置: Debug x64 ------
1>Generate moc_predefs.h
2>Generate moc_predefs.h
3>Generate moc_predefs.h
4>Generate moc_predefs.h
5>Generate moc_predefs.h
6>Generate moc_predefs.h
4>mapoptdlg.cpp
1>rtkconv_qt.vcxproj -> C:\Users\user\Desktop\nowdays\RTKLIB\app\qtapp\rtkconv_qt\debug\rtkconv_qt.exe
3>srctblbrows_qt.vcxproj -> C:\Users\user\Desktop\nowdays\RTKLIB\app\qtapp\srctblbrows_qt\debug\srctblbrows_qt.exe
4>C:\ProgramData\Qt\6.6.2\msvc2019_64\include\QtGui\qwindowdefs_win.h(37,1): error C2371: “HINSTANCE”: 重定义；不同的基类型
4>(编译源文件“mapoptdlg.cpp”)
4>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\minwindef.h(250,1):
4>参见“HINSTANCE”的声明
4>C:\ProgramData\Qt\6.6.2\msvc2019_64\include\QtGui\qwindowdefs_win.h(43,1): error C2371: “HDC”: 重定义；不同的基类型
4>(编译源文件“mapoptdlg.cpp”)
4>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\windef.h(80,1):
4>参见“HDC”的声明
4>C:\ProgramData\Qt\6.6.2\msvc2019_64\include\QtGui\qwindowdefs_win.h(46,1): error C2371: “HWND”: 重定义；不同的基类型
4>(编译源文件“mapoptdlg.cpp”)
4>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\windef.h(39,1):
4>参见“HWND”的声明
4>C:\ProgramData\Qt\6.6.2\msvc2019_64\include\QtGui\qwindowdefs_win.h(49,1): error C2371: “HFONT”: 重定义；不同的基类型
4>(编译源文件“mapoptdlg.cpp”)
4>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\windef.h(86,1):
4>参见“HFONT”的声明
4>C:\ProgramData\Qt\6.6.2\msvc2019_64\include\QtGui\qwindowdefs_win.h(52,1): error C2371: “HPEN”: 重定义；不同的基类型
4>(编译源文件“mapoptdlg.cpp”)
4>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\windef.h(94,1):
4>参见“HPEN”的声明
4>C:\ProgramData\Qt\6.6.2\msvc2019_64\include\QtGui\qwindowdefs_win.h(55,1): error C2371: “HBRUSH”: 重定义；不同的基类型
4>(编译源文件“mapoptdlg.cpp”)
4>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\windef.h(74,1):
4>参见“HBRUSH”的声明
4>C:\ProgramData\Qt\6.6.2\msvc2019_64\include\QtGui\qwindowdefs_win.h(58,1): error C2371: “HBITMAP”: 重定义；不同的基类型
4>(编译源文件“mapoptdlg.cpp”)
4>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\windef.h(73,1):
4>参见“HBITMAP”的声明
4>C:\ProgramData\Qt\6.6.2\msvc2019_64\include\QtGui\qwindowdefs_win.h(61,1): error C2371: “HICON”: 重定义；不同的基类型
4>(编译源文件“mapoptdlg.cpp”)
4>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\windef.h(88,1):
4>参见“HICON”的声明
4>C:\ProgramData\Qt\6.6.2\msvc2019_64\include\QtGui\qwindowdefs_win.h(67,1): error C2371: “HPALETTE”: 重定义；不同的基类型
4>(编译源文件“mapoptdlg.cpp”)
4>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\windef.h(93,1):
4>参见“HPALETTE”的声明
4>C:\ProgramData\Qt\6.6.2\msvc2019_64\include\QtGui\qwindowdefs_win.h(70,1): error C2371: “HRGN”: 重定义；不同的基类型
4>(编译源文件“mapoptdlg.cpp”)
4>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\minwindef.h(252,1):
4>参见“HRGN”的声明
4>C:\ProgramData\Qt\6.6.2\msvc2019_64\include\QtGui\qwindowdefs_win.h(73,1): error C2371: “HMONITOR”: 重定义；不同的基类型
4>(编译源文件“mapoptdlg.cpp”)
4>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\windef.h(110,1):
4>参见“HMONITOR”的声明
4>C:\ProgramData\Qt\6.6.2\msvc2019_64\include\QtGui\qwindowdefs_win.h(76,1): error C2371: “HGLRC”: 重定义；不同的基类型
4>(编译源文件“mapoptdlg.cpp”)
4>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\windef.h(82,1):
4>参见“HGLRC”的声明
2>strsvr_qt.vcxproj -> C:\Users\user\Desktop\nowdays\RTKLIB\app\qtapp\strsvr_qt\debug\strsvr_qt.exe
5>rtkget_qt.vcxproj -> C:\Users\user\Desktop\nowdays\RTKLIB\app\qtapp\rtkget_qt\debug\rtkget_qt.exe
6>rtknavi_qt.vcxproj -> C:\Users\user\Desktop\nowdays\RTKLIB\app\qtapp\rtknavi_qt\debug\rtknavi_qt.exe
4>已完成生成项目“rtkplot_qt.vcxproj”的操作 - 失败。
========== 生成: 5 成功，1 失败，3 最新，0 已跳过 ==========
========== 生成 于 14:56 完成，耗时 14.986 秒 ==========

JensReimann, 2024-04-21:
The issue should be fixed with the latest commit in the qt_gui branch. 

If the error remains, could you please specify the compiler you use and set the language for the error messages to english to be able to trace the problem.

Thanks

Bzz2745, 2024-04-21:
first, fix this(line 147 miss the ';'):
[image]

Bzz2745, 2024-04-21:
secondly, rtkplot_qt\plotdta.cpp missing the include file locale.h due to use setlocale(LC_NUMERIC, "C");

JensReimann, 2024-04-21:
Line 147 of kmzconv.cpp should have a semicolon ";" if you're using the latest qt_gui branch.

I'll check the compilation with your compiler/Qt version soon. Thanks already in advance for pointing to this issue.

Bzz2745, 2024-04-21:
> Line 147 of kmzconv.cpp should have a semicolon ";" if you're using the latest qt_gui branch.
> 
> I'll check the compilation with your compiler/Qt version soon. Thanks already in advance for pointing to this issue.

The same issues appeared when I build with qt5.15.2 for mingw

JensReimann, 2024-05-01:
I checked the compilation with qt5.15.2 and mingw 8.1 on my system and it works fine. 

Can you please double check that you are using the latest version of the qt_gui branch? The changes are not yet in the demo5 branch.

JensReimann, 2024-05-26:
As there is no further response, I will close the issue and consider it as solved.

closed 2024-05-26 by JensReimann