rtklibexplorer/RTKLIB #210: ’lam_carr‘ Global variable that have been removed, but are still used in /src/rcv/comnav.c and tersus.c

Bzz2745, 2024-03-26:
The global variable has been removed on update 2020/11/30 in version 1.45, but in comnav.c, there is also many bugs appeared in comnav.c when I building the consapp. What shuld I do or what can I help?

rtklibexplorer, 2024-03-27:
Tersus.c and Comnav.c are no longer used in any of the standard builds and have not been supported for some time, but may be still referenced in some legacy build files.  Which build file are you using when you see errors.

Bzz2745, 2024-03-27:
I'm trying to build consapp. And I want to know if you have any interested in provide a Cmake file for the project which is really necessary while using in vscode to build and debug both on windows and linux.发自我的手机