rtklibexplorer/RTKLIB #255: Error converting from dms to degrees or ECEF in rtkpost_qt

t2rtech, 2024-04-27:
When using the Qt version of RTKPOST_QT, an error in conversion occurs when changing the input coordinate type from DMS to degrees or ECEF in the Positioning tab of the Options Dialog.

For example, when inputting latitude, longitude, and height values in degrees/minutes/seconds (DMS) format:

Latitude: -23.206632745293
Longitude: -47.148947742523
Height: 746.78767

And then changing to DMS format:

Latitude: -23 12 23.877883
Longitude: -47 8 56.211873
Height: 746.7877

Upon choosing degrees or ECEF format again, you'll notice that the longitude coordinate was set to zero.

The problem occurs due to duplicated spaces between "-47" and "8" or other instances of multiple spaces between the numbers.

I fixed the code by adding Qt::SkipEmptyParts in the split function, like this, in the function in postopt.cpp file:

**void MainWindow::GetPos(int type, QLineEdit **edit, double *pos)**

```
    if (type == 1) { /* lat/lon/height dms/m */
        QStringList tokens = edit0_Text.split(' ', Qt::SkipEmptyParts);

        if (tokens.size() == 3)
            for (int i = 0; i < 3; i++) dms1[i] = tokens.at(i).toDouble();

        tokens = edit1_Text.split(' '**, Qt::SkipEmptyParts)**;

        if (tokens.size() == 3)
            for (int i = 0; i < 3; i++) dms2[i] = tokens.at(i).toDouble();

```

Probably some adjust can be done in function:

void MainWindow::SetPos(int type, QLineEdit **edit, double *pos)

to avoid insert a duplicated space in the convertion (or put a leading 0), to maintain the format

JensReimann, 2024-04-29:
Thanks a lot for reporting this issue.

I will probably take the chance and further improve the handling of the various input formats in a upcoming commit.

JensReimann, 2024-05-01:
The latest commit in the qt_gui branch fixes the issue. 

Now, the user input is validated using regular expressions and the units are show, too. This may help users to correctly fill the input.

Please test!

closed 2024-05-24 by JensReimann