rtklibexplorer/RTKLIB #262: Antenna type not shown in .pos file

LabGeomatica, 2024-05-08:
Good morning,
I'm processing data from a Topcon NET-G3A receiver with TPSPG_A1 antenna. 
In the File tab of Options I've put the igs14.atx file and I've specified the antenna name (according to the igs14.atx file) in the Antenna Type in Position tab.
After processing, I've noticed that in the .pos file, the specified antenna model is not present in the line corresponding to "antenna1".
I'm wondering if the antenna model is then considered in the processing and what I'm doing wrong.
I attach an extract of the output I obtain and the configuration file.
Thank you for your kind attention.

[image]

[configuration_file.txt](https://github.com/rtklibexplorer/RTKLIB/files/15246533/configuration_file.txt)

rtklibexplorer, 2024-05-08:
There are two antenna file fields in the Options File tab, one for the satellites, and one for the receivers.  From your config file, it looks like you have set the satellite antenna file but not the receiver antenna file.  Normally, you can set both of these to the same file.

LabGeomatica, 2024-05-09:
Oh, shame on me! 
Thank you for your time and answer!

closed 2024-05-09 by rtklibexplorer