rtklibexplorer/RTKLIB #265: rtknavi_qt/rtkrcv: invalid option value inpstr2-type

escuta, 2024-05-09:
For my Base Station input stream, I am unable to set it as  "NTRIP Client" from the conf file. When I create a conf file in rtknavi_qt, the Base Station input stream gets saved as the following:

`inpstr2-type       =6`

When the conf file is reloaded, the following error results:

`invalid option value inpstr2-type (/home/me/rtk/ntrip.conf:131)`

The conf file itself shows the following index for inpstr2-type in the comments:

`# (0:off,1:serial,2:file,3:tcpsvr,4:tcpcli,7:ntripcli,8:ftp,9:http)`

But if I edit the file and set 

`inpstr2-type       =7`

or 

`inpstr2-type       =ntripcli`

There is no error, however the stream type for the Base Station does not get set in the rtknavi_qt GUI and the stream is not received. If I manually set the stream type to "NTRIP Client" in the GUI, the base station stream is received correctly. rtkrcv also does not work when the config file is loaded.

I have also tried using other index values such as 5 and 6, but nothing sets the input stream correctly as "NTRIP Client"

Please help!

rtklibexplorer, 2024-05-11:
Yes, some significant improvements coming soon for the QT GUIs from Jens Reimann, the original author of them.  Still a few minor issues to resolve, but you can try them now on the qt_gui branch.  The install process isn't quite sorted out yet so you may need to manually set the LD_LIBRARY_PATH environment variable to point to the shared libraries.

Currently the dev branch is just for internal development and doesn't have any additional features.

JensReimann, 2024-05-13:
Dear escuta, 
the issue should be fixed with the latest commit to the qt_gui branch. 
If you find similar issue feel free to report them here.

JensReimann, 2024-06-09:
closed due to inactivity

closed 2024-06-09 by JensReimann