rtklibexplorer/RTKLIB #267: The b34i version cannot be fixed when using the bds system for moving base calculation

joseph-chunhui, 2024-05-11:
The following is the solution related configuration and solution result. It is obvious that the solution result has failed.
[image]
[image]
[image]
In the data, the base station signal undergoes occlusion once every period of time and then recovers to a good signal.
[bds.zip](https://github.com/rtklibexplorer/RTKLIB/files/15281382/bds.zip)

rtklibexplorer, 2024-05-12:
I am able to duplicate your issue.  The solution fixes OK with integer ambiguity for Beidou disabled ( "Integer Ambiguity Res (BDS)"=OFF in Setting2 tab) but not with this enabled.  I need to take a closer look to see if this is an issue with your Beidou receiver data or an issue with the RTKLIB code but in the meantime you should be fine with this option disabled.  Beidou integer ambiguity resolution does work fine in RTKLIB in general with other receivers.  

By the way, you have L1/L2/L5 observations for both rover and base so you can set the Frequencies option to L1/L2/L5 for better results.

joseph-chunhui, 2024-05-15:
ok！The bds system also fails to fix ambiguity when using full-system solutions in rtklib.
What algorithm is used to solve L1/L2/L5 RTK in RTKLIB? Is it EWL+WL mode?

rtklibexplorer, 2024-05-15:
I just realized from your screenshot that you are trying to solve for BDS only.   Because RTKLIB was originally written for GPS only, in normal circumstances you should always have GPS selected for a RTK or PPK solution.  Also, for PPK solutions, you will generally get a better solution with "Filter Type" set to combined rather than forward.
 
RTKLIB does not not use wide-lane or extra wide-lane combinations for ambiguity resolution, only uncombined signals.

joseph-chunhui, 2024-05-16:
Yes, I would like to confirm whether rtklib can be solved using only BDS system for RTK?
Logically, as long as the signal supports, the RTK solution using only the BDS system and the RTK solution using only the GPS system should have the same result.

rtklibexplorer, 2024-05-19:
Yes, in theory, you should be able to get an RTK solution with just BDS
satellites, but RTKLIB has some dependencies on the GPS satellites due to
its GPS-only origins, so you do need to include them.

joseph-chunhui, 2024-05-20:
But in fact, I use GPS+BDS at the same time, BDS satellite can not fix the whole week blur, only GPS satellite can fix the whole week blur, what is the reason?