rtklibexplorer/RTKLIB #291: qt_branch: Input Streams -> Stream Options : Port always 0 for TCP Client

dwrobel, 2024-05-20:
rtknavi_qt revision 68a355b on Fedora 39

Reproducible path:
1. launch rtknavi_qt
2.  press "I" button (Define input streams), then activate "(1) Rover" with Stream Type: "TCP Client", Stream Options: { Server Address: localhost, Port: 12345}, then press "OK" button.
3. Press again "I" button then press "..." button for under "Stream Options".

Expected result: both "Server Address" and "Port" should be preserved.
Current result: "Server Address" is preserved _(localhost)_ but "Port" is always 0 _(instead of 12345)_.

JensReimann, 2024-05-20:
Should be fixed (together with same other issues) with the latest commit to the qt_gui branch. Please give it a try.

dwrobel, 2024-05-21:
Tested using 37ce585 and it works fine. Thanks for the fix.

closed 2024-05-21 by dwrobel