rtklibexplorer/RTKLIB #302: The b34i version has buffer overrun issues in NTRIP

DavidKelleySCSC, 2024-05-28:
I am seeing **QUITE A LOT** of buffer overrun and garbled message content when RTKLIB devices using b34i (typically rtknavi.exe I presume) try to connect to the SNIP NTRIP Caster at RTK2go.com (or 3.143.243.81:2101).  We have been having some problems the past couple of weeks on that machine with a few users jamming it with rapid repeated bad connections (and hence harming another thousand who are using it correctly).  But when the machine resets (going offline for about 2~3 minutes), we then get a great number of RTKLIB derived user devices (in this case they are all using **b34i**) who then send in garbled data until they are power cycled (which seems to be the only cure).  And sometime the problem device will send in very long repeating nonsense in the reset of the NTRIP header as well.  The garbled data seems to be part of that Caster reply, words like "ICY OK" etc. This really seems like a buffer management issue in a string somewhere as the device will repeat some of the NTRIP reply text and send that back in strange places such as the mountPt name.  To be fair, I have seen this occasionally for years in RTKLIB (and never been able to track it down), but now we are seeing it quite a lot to point of being it is becoming a major defect.    Below is a redacted extract of a few IPs the machine blocked due to this today.  Note the weird base station names (mountPt names).  Will provide any help that I can to get to the bottom of this and fix it.  

[image]

DavidKelleySCSC, 2024-05-29:
Just a morning update. Seeing about four dozen such users this morning on the RTK2go NTRIP Caster.  If this is affecting you (it affects NTRIP _Servers_ not NTRIP _Clients_), please reset (power cycle) your device to fix it.  And please NEVER re-connect more frequently than once per second (we are getting 450+ connections a minute this morning form these garbled devices).

rtklibexplorer, 2024-05-29:
Sorry to hear the latest release is causing issues.  I have added a prominent note at the top of the release description on Github with a link to the issue and a request for users to use the b34h release for any NTRIP streaming until this is resolved.

Fortunately there has been a recent and significant effort by OurAirQuality to submit pull requests to clean up many of the outstanding out-of-bound and buffer overflow errors in the code so it is possible that this is already resolved in the latest code, or will be when I finish accepting these requests.  I will also review the differences between b34h and b34i related to streaming to see if I can see anything obvious.

Do you have any suggestions on the best way that I can duplicate this problem for testing and verification purposes?

rtklibexplorer, 2024-05-29:
OK, I have merged all outstanding out-of-bound and buffer overflow pull requests into the demo5 branch so if anyone is in a position to validate the current source code it would be an interesting experiment.  I'm happy to do it myself if anyone can give me instructions on how to best duplicate the problem.

I've compared the streaming code between b34h and b34i releases and there is almost no differences between them.  It's also curious that the b34i release came out at the beginning of January but the issues only seem to have cropped up in the last couple of weeks.  Is it possible that there has been a recent change in behavior on the RTK2GO side that could be aggravating this bug?

If it is not possible to reliably duplicate this issue, the best path might be for me to create a new release with the latest code and see if this eliminates the problem.  Thoughts?

ourairquality, 2024-05-30:
It does look like memory corruption, use after free, etc, something that has a good chance of being caught by a hardened build.

It's odd that there have been no reports from users, surely some are getting frustrated?

Is there some forum to reach out to your users to ask them to make reports. Perhaps one of them would be prepared to run some development builds to help.

Limited local testing on hardened linux builds has not reproduced this issue, with the client talking to its own server or to the RTK2go.com servers.

A network capture might assist, might indicate the state of the client leading up to this. Might there also be some way differentiate the client OS by a network capture.

The code does not look all that solid!

Stefal, 2024-06-03:
A new RTKBase release will be available in the coming weeks. Perhaps I should wait. @rtklibexplorer, do you plan to release a b34j soon?

rtklibexplorer, 2024-06-03:
Yes, I hope to put out a new release this week.

Stefal, 2024-06-04:
@DavidKelleySCSC Did you notice if some of these sources are using RTKBase? Once these problematic sources have been reconnected correctly they should send the rtkbase release number in the RTCM message 1033.

rtklibexplorer, 2024-06-08:
The b34j code release is now available.  This has many fixes for potential out-of-bound and buffer overflows but no direct verification that it has solved this issue.  Please let me know if you see any similar issues with this release.

DavidKelleySCSC, 2024-06-08:
Everyone, thank you do much for the efforts to fine and resolve this.  Just at the moment (June 9th ~3:00 PM UTC time) I do not yet show any b34j code users yet on the RTK2go machine.  I will keep an eye on this and report back once we see some usage. 
```
NTRIP RTKLIB/demo5_b34g    (100 entries) 
NTRIP RTKLIB/demo5_b34h    (5 entries) 
NTRIP RTKLIB/demo5_b34i    (132 entries) 
```

rtklibexplorer, 2024-08-07:
Haven't had any reports of issues with the latest code so I'm hopeful that the recent changes resolved the problem.

closed 2024-08-07 by rtklibexplorer

DavidKelleySCSC, 2024-08-09:
Two month later,... We are now showing quite a few 'j' release user connections on that Caster, so far no buffer issues I can trace back to those. So cautiously optimistic. 
   NTRIP RTKLIB/demo5_b34i    (103 entries)  (plus another 80+ that are older)
   NTRIP RTKLIB/demo5_b34j    (47 entries)