rtklibexplorer/RTKLIB #423: "RTCM/Raw Antenna Position" Base station position does not work in rtknavi_qt

escuta, 2024-07-24:
Hello,

I recently came back to my RTKLIB project after several months and rebuilt the quicktime and command line apps. I'm using a local NTRIP server for my base station and always set its position to "RTCM/Raw Antenna Position". With the latest origin/HEAD build, I am unable to get solutions for the rover's position in rtknavi_qt. 

The Position tab of the gui does read and write to the config file correctly with: "ant2-postype       =rtcm". It doesn't however seem to have any effect. The rtknavi_qt gives the warning/error: "Initial base station position error"

If I jump back to a commit on 1 April, "RTCM/Raw Antenna Position" works correctly and solutions are produced.

Thanks,

JensReimann, 2024-07-27:
Thanks for reporting this bug, escuta.

Please try the latest git version which hopefully fixes this bug. There was a typo introduced causing the RTCM option to be applied to the rover antenna instead to the base station.  I used this issue to also cleaned-up this code part as bit.

escuta, 2024-07-27:
Thanks a lot Jens. Yes, that seems to be working. I'll be able to check it more thoroughly during the week. All the best,

JensReimann, 2024-07-27:
Perfect. Give me a hint, if the problem is yet not fully solved.