rtklibexplorer/RTKLIB #446: Unicore Message Length Error

bemasher, 2024-08-11:
In rtknavi, b34k, with a UM980, the unicore decoder is rejecting all messages with a length error.

Receiver commands:
```
UNLOG

OBSVMB 1
GPSEPHB ONCHANGED
GLOEPHB ONCHANGED
GALEPHB ONCHANGED
```

Receiver configuration:
```
$command,VERSIONA,response: OK*45
#VERSIONA,96,GPS,FINE,2327,2682000,0,0,18,600;"UM980","R4.10Build11833","HRPT00-S10C-P","2310415000001-MD22B1225041718","ff3bde96eeb6113d","2023/11/24"*8aa0e1e2
$command,CONFIG,response: OK*54
$CONFIG,ANTENNA,CONFIG ANTENNA POWERON*7A
$CONFIG,NMEAVERSION,CONFIG NMEAVERSION V410*47
$CONFIG,RTK,CONFIG RTK TIMEOUT 120*6C
$CONFIG,RTK,CONFIG RTK RELIABILITY 3 1*76
$CONFIG,PPP,CONFIG PPP TIMEOUT 120*6C
$CONFIG,PPP,CONFIG PPP ENABLE E6-HAS*01
$CONFIG,PPP,CONFIG PPP DATUM WGS84*02
$CONFIG,DGPS,CONFIG DGPS TIMEOUT 300*6C
$CONFIG,RTCMB1CB2A,CONFIG RTCMB1CB2A ENABLE*25
$CONFIG,ANTENNADELTAHEN,CONFIG ANTENNADELTAHEN 0.0000 0.0000 0.0000*3A
$CONFIG,PPS,CONFIG PPS ENABLE GPS POSITIVE 500000 1000 0 0*6E
$CONFIG,SIGNALGROUP,CONFIG SIGNALGROUP 2*16
$CONFIG,ANTIJAM,CONFIG ANTIJAM AUTO*2B
$CONFIG,AGNSS,CONFIG AGNSS DISABLE*70
$CONFIG,BASEOBSFILTER,CONFIG BASEOBSFILTER DISABLE*70
$CONFIG,COM1,CONFIG COM1 460800*2E
$CONFIG,COM2,CONFIG COM2 115200*23
$CONFIG,COM3,CONFIG COM3 115200*23
```

Trace level 2 output:
```
2 unicore length error: len=40984
2 unicore length error: len=40984
2 unicore length error: len=40984
...
```

I believe the issue is the offset the message length is being read from:

https://github.com/rtklibexplorer/RTKLIB/blob/55a0f2c742a605a3d27a90b17556b2cfdf98ad50/src/rcv/unicore.c#L881

I wrote a program to test this, and in the messages returned from my receiver the uint16 value at bytes 8:10 is 40960 - 24 = 40984.

I believe the message length should be at bytes 6-8.

ourairquality, 2024-08-11:
Yes the documentation says the length is at offset 6, table 7-49. See the "Unicore N4 Products Commands and Logs Reference Book R1.4" which might help make progress further progress.There is apparently a Unicore converter (binary not source) and comparing the output with RTKLib might help make progress.

rtklibexplorer, 2024-08-12:
I have tested the Unicore binary parsing feature in the b34k code using a binary file containing raw observations from a Unicore 982 receiver.  This worked fine, so the issue does not affect all Unicore binaries.  As OurAirQuality suggested, it would be worthwhile to confirm that the Unicore converter can successfully parse the file.

geofis, 2024-08-12:
Hello,

Thanks for these improvements.

I tested with a dataset mentioned here (it doesn't include navigation data):

https://github.com/Stefal/rtkbase/issues/351#issuecomment-1970259052

Indeed, the data collected with OBSVMB can be successfully converted with convbin:
```
wine ./convbin.exe obsvmb.unc 
input file  : obsvmb.unc (Unicore)
->rinex obs : obsvmb.obs
->rinex nav : obsvmb.nav
->sbas log  : obsvmb.sbs
scanning: 2024/02/28 23:30:03 GREC
2024/02/28 23:28:38-02/28 23:30:05: O=88 E=1
```

The RINEX header looks like this:

```
G    8 C1C L1C C1L L1L C2L L2L C5Q L5Q                      SYS / # / OBS TYPES 
R    4 C1C L1C C2C L2C                                      SYS / # / OBS TYPES 
E    8 C1C L1C C7Q L7Q C5Q L5Q C6C L6C                      SYS / # / OBS TYPES 
C   12 C2I L2I C1P L1P C7I L7I C7P L7P C5P L5P C6I L6I      SYS / # / OBS TYPES 
```

The Z signal is not recovered in the conversion, but interestingly, the L1C signal is.

When converting with the converter, the Z signal is recovered, but there's no trace of L1C. Other differences can be observed in the recovered signals across other constellations:

```
G   16 C1C L1C D1C S1C C2L L2L D2L S2L C2W L2W D2W S2W C5Q  SYS / # / OBS TYPES 
       L5Q D5Q S5Q                                          SYS / # / OBS TYPES 
R    8 C1C L1C D1C S1C C2C L2C D2C S2C                      SYS / # / OBS TYPES 
C   24 C2I L2I D2I S2I C7I L7I D7I S7I C6I L6I D6I S6I C1P  SYS / # / OBS TYPES 
       L1P D1P S1P C5P L5P D5P S5P C7D L7D D7D S7D          SYS / # / OBS TYPES 
E   12 C1C L1C D1C S1C C5Q L5Q D5Q S5Q C7Q L7Q D7Q S7Q      SYS / # / OBS TYPES 
```

I imagine it might be related to the footnote in Table 7-55 of the "NebulasIV High Precision Products. Commands and Logs" document, which states: "5 When the value of Bit 26 is 1, the L2P (Y) in Bit 25 is actually the L2C signal."

Finally, I cannot achieve a proper conversion with the data collected using RANGEB:

```
wine ./convbin.exe rangeb.unc 
input file  : rangeb.unc (Unicore)
->rinex obs : rangeb.obs
->rinex nav : rangeb.nav
->sbas log  : rangeb.sbs

scanning: 1970/01/01 00:00:00 
:
```

Best regards.

ourairquality, 2024-08-12:
input_unicore() currently uses an offset of 8 for the length vs input_unicoref() which use an offset of 6, so this might not show up when decoding a file. Try changing the offset in  input_unicore()  and see if it get further, and perhaps check the logic there.

Making binary datasets available can help. The Unicore 980 appears to support more signals so a dataset from it with all the signals output would help, and signal group 2 would appear to have the GPS L2P(Y). Also might the RTCMSWITCHL2CL2P command change the binary output or the interpretation and might it be related to that bit 26 flag?

bemasher, 2024-08-14:
The UM980 can respond with one of two binary formats that are similar, but not interchangeable. Some commands return Novatel OEM7 compatible messages (sync AA4412), and some return Unicore's flavor (sync AA44B5). Novatel's header is slightly longer and the message length starts at byte 8, while Unicore's starts at byte 6.

I've attached a csv file of all the commands my UM980 responded "OK" to, and the sync word it produced, if any.

[commands.csv](https://github.com/user-attachments/files/16606869/commands.csv)

bemasher, 2024-08-16:
I will work on making a binary dataset for the UM980 this weekend.

bemasher, 2024-08-17:
I've recorded examples for all the binary messages my UM980 would respond to.

The `BDSQIONUTCB` message appears to have a malformed header, but the CRC is valid.

`um980.csv` includes command, sync, offset, length, message length, and message ID for each binary message in `um980.bin`.

[um980.tar.gz](https://github.com/user-attachments/files/16643279/um980.tar.gz)

geofis, 2024-08-17:
Hi,

Thank you, @bemasher.

I tested converting your data to RINEX using convbin from demo5 and Unicore's converter (both attached, the one with the .obs extension was generated by convbin, the other one by converter).

I noticed some differences. First, for GPS, the Z-tracking signal (*2W) is captured by Unicore's converter but is omitted by convbin (see, for example, SV G04 from the first epoch). In GLONASS, convbin manages to recover G3, but the converter does not (see R05, first epoch). As for Galileo, the converter doesn't extract E6, whereas convbin does. Finally, by default, convbin omits BeiDou, but the converter successfully extracts observables from that constellation.

Regards,

[um980.zip](https://github.com/user-attachments/files/16643439/um980.zip)

bemasher, 2024-09-03:
> @geofis 
> I imagine it might be related to the footnote in Table 7-55 of the "NebulasIV High Precision Products. Commands and Logs" document, which states: "5 When the value of Bit 26 is 1, the L2P (Y) in Bit 25 is actually the L2C signal."

I had a look through the Unicore code and it is correctly handling those footnotes for swapping L2C and L2P(Y) in GPS and QZSS observations.

bemasher, 2024-09-03:
I've corrected the length issue in PR #466.

commit 3bf4841f (2024-10-02): Merge pull request #466 from bemasher/demo5

Unicore: Fix message length in input_unicore. Fixes #446
closed 2024-10-02 by rtklibexplorer