rtklibexplorer/RTKLIB #447: xxx is not a member of a Qt Class

WangJitaoNAV, 2024-08-12:
I encountered problems when trying to compile the rtklib_qt gui version on ubuntu 20.04. Most of them are incompatible with the QT APIs.  For example:
```
browsmain.cpp:113:40: error: ‘textActivated’ is not a member of ‘QComboBox’
  113 |     connect(ui->cBAddress, &QComboBox::textActivated, this, &MainForm::getTable);
      |                 
```
```
../appcmn_qt/fileoptdlg.cpp:86:72: error: ‘class QStringView’ has no member named ‘toDouble’
   86 |             if (token.contains("x")) speed = QStringView{token}.mid(1).toDouble();
      |                                                                        ^~~~~~~~
../appcmn_qt/fileoptdlg.cpp:87:76: error: ‘class QStringView’ has no member named ‘toInt’
   87 |             if (token.contains('P')) size_fpos = QStringView{token}.mid(2).toInt();
```
My environment is ubuntu 20.04 with libqt5-default. I wonder how could I fix this problem. Any help would be appreciated!

WangJitaoNAV, 2024-08-12:
ps: RTKLIB Version is the newest with b34k

WangJitaoNAV, 2024-08-13:
Thank you for your prompt response! 

This is indeed a compatibility issue caused by the outdated version of Qt5. My previous version of Qt5 was 5.12.8, which is the default version corresponding to Ubuntu 20.04. After updating to 5.15.2, the compilation was successful, and I am now able to use RTKLIB GUI on my Ubuntu. 

Thank you again for your contribution.

closed 2024-08-13 by WangJitaoNAV

harlock974, 2024-08-19:
I had the same issue with Linux Mint 20.2 (based on Ubuntu 20.04 LTS Focal Fossa).
You can't update Qt from repositories, but you can use a PPA :
```bash
sudo add-apt-repository ppa:beineri/opt-qt-5.15.2-focal
sudo apt update
sudo apt install qt515base qt515serialport
export PATH="/opt/qt515/bin/:$PATH"
qmake --version # check if it is 5.15.2
cd RTKLIB-b34k/app/qtapp/
qmake
make
./install_qtapp
```
You need to use Qt version 5.15.2. Version 5.15.0 brings the same error messages reported by [WangJitaoNAV](https://github.com/WangJitaoNAV).
Despite having installed the RTKLIB b34k version from sources this way, the window title from the GUI and the about window show b34j. Is it a mistake or does it mean that the Linux version had not been actually updated ?

rtklibexplorer, 2024-08-19:
My mistake on the version number for the b34k release, the initial release files did not have the version number changed, I updated them on 8/13.