rtklibexplorer/RTKLIB #476: using convbin.exe to convert rtcm3 into rinex, the GLO is missing L1,L2 data

fuyongjia, 2024-09-30:
[image]
The RTCM data includes broadcast ephemeris.

ourairquality, 2024-09-30:
Do not see this, in demo5. Perhaps run convbin with tracing enabled and check on the messages being received, its it 1012 or the MSM messages, just to be sure that the data is in the stream, and it might narrow it down.

fuyongjia, 2024-10-09:
When I use rtkconv.exe, I manually configured the FCN table, and the converted rinex result file is complete.
[image]
[image]

ourairquality, 2024-10-09:
Yes it needs the frequency to be able to produce the carrier phase. This appears to be included in the MSM5 and MSM7 messages, or be obtained from the nav data, or as you noted can be prefilled in the nav.glo_fcn from the glofcn option but this option is not available on the command line app convbin and perhaps that support needs to be added.

ourairquality, 2024-10-09:
The was reproducible using RTCM3 MSM6 which lacks the GLONASS fcn, and initializing the fcn overcame this.

https://github.com/rtklibexplorer/RTKLIB/pull/485 adds support to the convbin command line tool, e.g.

`convbin --glofcn "1,-4,5,6,1,-4,5,6,-2,-7,0,-1,-2,-7,0,-1,4,-3,3,2,4,-3,3,2" ...`

closed 2024-10-20 by rtklibexplorer