rtklibexplorer/RTKLIB #534: the setting of rinex file type

schumipush, 2024-11-21:
I suspect the "buff+21" is wrong.

For a rinex 3.04/3.05 file, the variable "type" would be wrong.

https://github.com/rtklibexplorer/RTKLIB/blob/aa79681bfec8d284fb10cb1ec24dc550c3bde3c3/src/rinex.c#L680

ourairquality, 2024-11-21:
For a 3.04 clock file the type does appear to be at buff+21.

But see a 'M' at buff+42 in a 3.04 clock file and if this is the sat system then the code looking for this at buff+40 might be wrong - but the sys does not appear to be used reading the clock file.

schumipush, 2024-11-22:
Thanks a lot! @ourairquality 

I have found two examples as follows:

--------------------
3.04 (buff+21)
3.04                 C                                           RINEX VERSION / TYPE
-------------------
2.00 (buff+20)
     2.00           CLOCK DATA                              RINEX VERSION / TYPE

closed 2024-11-22 by schumipush