rtklibexplorer/RTKLIB #548: Unicore reject GLO L2 freq

juffe84, 2024-12-19:
Hello.

Im trying to stream base station data to the caster and with strsvr im do conversion from unicore to legacy rtcm3.0.
I cant get L2 signals with conversion. What is problem? I think so that 1075 conversion works fine, but I cant test it, because I dont have payed version of SNIP.
My settings:
CONFIG SIGNALGROUP 2
CONFIG UNDULATION 0
OBSVMB 0.5
GPSEPHB 5
GLOEPHB 5
BDSEPHB 5
GALEPHB 5
QZSSEPHB 5
IRNSSEPHB 5
BD3EPHB 5
[image]

juffe84, 2024-12-20:
Little bit of researching and I can get L2C code with option -RL2C. Cant get L2P code.

ourairquality, 2024-12-20:
The current unicore driver does not include an entry for GLONASS L2P as the documentation has none. Was L2P expected to be supported?

ourairquality, 2024-12-21:
Thank you for the link to the v1.6 manual. There does not appear to be any G2P support in the observation messages. It does seem a little odd that RTCMSTATUS has a G2P entry, but perhaps it is an allocation for some variant. Did note support for two new QZSS signals and shall submit a PR to add decoder support for those.

rtklibexplorer, 2025-02-04:
QZSS signal support added with PR #553