rtklibexplorer/RTKLIB #554: Windows 11 MinGW 11.2.0 and Qt 6.5.3 Compilation Error

liusheng2020, 2024-12-23:
When compiling demo5-b34k using MinGW 11.2.0 and Qt 6.5.3 on Windows 11, a typedef conflict error occurs involving such as the types HRGN and HMONITOR. This issue prevents the project from being compiled successfully.
[image]

I’m unsure how to resolve this issue. Should I consider using a different Qt version or perhaps a different version of the rtklib-demo5? This problem has been troubling me for a long time, and I would greatly appreciate any guidance or advice.

[image]

JensReimann, 2024-12-26:
Hi liusheng2020,
unfortunately, I have no Windows computer on my hand to debug the problem myself right now, but I experienced similar compilation errors before. 

Please, look for the location where the compilation originally origins from. It often helped to change the order of the include files while having the Qt includes (those which in turn include the minwindef.h file) at first and all other files, especially the includes of rtklib itself, later.

If you manage to resolve the compilation problem, feel free to post a merge request.

Next week, I have access to a Windows computer again to start debugging.

JensReimann, 2025-01-02:
Dear liusheng2020,
a patch for the compilation problem has been committed. Please check the latest git version.

closed 2025-01-02 by JensReimann

liusheng2020, 2025-01-02:
Thank you very much.  With the updated code，I success.

liusheng2020, 2025-01-02:
Thank you very much. With the updated code，I success compiling demo5-b34k using MinGW 11.2.0 and Qt 6.5.3 on Windows 11> Dear liusheng2020,
> a patch for the compilation problem has been committed. Please check the latest git version.