rtklibexplorer/RTKLIB #556: solution for the case of "epoch flag == 5"

schumipush, 2025-01-07:
Is the symbol "<=" correct?

To my understanding, if (fabs(timediff(data[0].time,time1)-dtime1)>=DTTOL), then the new ely read epoch is valid. So why do you want to discard the current epoch data and read the next epoch in turn?

https://github.com/rtklibexplorer/RTKLIB/blob/61ca04f254e6486fc80ac6fa2f7a3802e466a657/src/rinex.c#L1134

ourairquality, 2025-01-09:
Fwiw this code is specific to time events, when flag is 5, so does not affect common uses. At a guess it was to assign the event to a set of observation data, but it needs some work. It does not appear that the events are used for anything. Perhaps the only code paths tested were the output paths, to get events from the receiver data into rinex files.

schumipush, 2025-01-10:
@ourairquality Thanks for your response!  I think there might be an underlying assumption that with a flag of 5, the observations in next epoch might be  unhealthy or of poor quality. Subsequently, if the next epoch does not occur at the established interval, the epoch is considered unavailable and skips to the next epoch. What do you think?

rtklibexplorer, 2025-02-04:
I believe Our Air Quality is correct and that this is simply a misunderstanding of the flag=5 functionality.

closed 2025-02-04 by rtklibexplorer