rtklibexplorer/RTKLIB #557: str2str, multiple rtcm3 outputs with different messagetypes

roska7, 2025-01-07:
How can I stream multiple outputs with different rtcm message types? I think so that only latter -msg is used.

My string is like that and I get same msm4 format to both of mount points.
./str2str -in serial://ttyUSB0#unicore -out ntrips://:pass@localhost:2101/RTCM3#rtcm3 -msg "1004(1),1005(5),1008(5),1012(1),1019(5),1020(5)" -out ntrips://:pass@localhost:2101/RTCMMSM#rtcm3 -msg "1006(5),1008(5),1033(5),1074(1),1084(1),1094(1),1124(1),1230(5)" -p 0.000000 0.000000 29.5987 -a ADVNULLANTENNA -opt -RL2C

Also wrong information in --help     unc          : Unicore (only in), should be     unicore         : Unicore (only in)

ourairquality, 2025-01-08:
This is not currently supported in the console app, but appears to be in the gui apps. Might you try https://github.com/rtklibexplorer/RTKLIB/pull/558 and place the -msg option before the respective -out option.

roska7, 2025-01-08:
Thank you, your solution works like a charm. Is maximum output streams 3? Can we easily get more streams?

ourairquality, 2025-01-09:
rtklib.h appears to define 16 input / output streams for a stream server, and str2str.c has MAXSTR of 5 streams. So str2str  should support 4 output streams by default. It appears that MAXSTR could be increased to 16 in str2str.c to give 15 output streams, and beyond that the allocations for strsvr_t in rtklib.h would need to be increased. Would also need to extend any  initializers in str2str.c that depend on the number of streams. Might want to add more -c[n] options for the new streams, or generalise that code. Might be some other issues, some baked in constants that need patching.

rtklibexplorer, 2025-02-04:
Resolved with commit of PR#558

closed 2025-02-04 by rtklibexplorer