rtklibexplorer/RTKLIB #562: Could You Please Enhance demo5's Flexibility by Adding CMakeLists.txt?

liusheng2020, 2025-01-10:
Thank you for your wonderful work! I have successfully set up debugging for rtklib-demo5 in VS2022 by opening the rtklib-demo5/app/consapp/XXX/msc/msc.sln file.

However, I am unsure how to debug it using the existing makefile. Additionally, I noticed that the current rtklib-demo5/app/consapp/XXX/gcc/makefile is designed for Ubuntu (Linux) and is not directly applicable to Windows.

CMake can lower the learning curve and simplify the compilation and usage of RTKLIB (demo5), especially for beginners. Could you please add a CMakeLists.txt file so that we can more easily debug consapp on both Ubuntu (Linux) and Windows? With CMakeLists.txt, it would be possible to automatically generate the appropriate makefile for different platforms, increasing the flexibility of the project.

I previously forked a CMake version of demo5 that works well (I have successfully tested it on Ubuntu using VSCode), but the author is no longer actively maintaining it, and more importantly, its folder structure is not synchronized with the latest demo5, so it cannot be directly used in this repository. Here is the project link for your reference: https://github.com/liusheng2020/demo5-cmake

Thank you!

JensReimann, 2025-01-19:
Please have a look at the latest commit "Initial support for CMake". It is optional, only preliminary and not yet tested with Visual Studio (but the Visual C compiler).
Feedback is welcome.

shegnauer, 2025-01-31:
See my comment on that latest commit. It breaks compilation with Embarcadero 11 + 12.

JensReimann, 2025-02-01:
The problem is, that CMake wil generate new makefiles which overwrite the manually created once. 
This is a general problem and hard to solve.

liusheng2020, 2025-02-04:
With the commit titled "Initial support for CMake," I am now able to successfully compile this project, including both the CUI (Command-Line Interface) and the QT GUI (Graphical User Interface), on both Linux and Windows platforms. However, I am encountering difficulties when trying to integrate Embarcadero into the CMakeLists.txt, even though the project compiles successfully using Embarcadero C++ Builder 12 directly.

Despite setting the following configurations in the CMakeLists.txt:
```
set(CMAKE_C_COMPILER "D:/Program Files (x86)/Embarcadero/Studio/23.0/bin/bcc32c.exe")
set(CMAKE_CXX_COMPILER "D:/Program Files (x86)/Embarcadero/Studio/23.0/bin/bcc32.exe")
set(CMAKE_CXX_COMPILER_ID  "Embarcadero")
```

the build process still fails. Here is the error message I encountered:
[image]

I have found a potentially useful resource that might help address this issue:
https://docwiki.embarcadero.com/RADStudio/Sydney/en/Using_CMake_with_C%2B%2B_Builder

ourairquality, 2025-04-03:
The IERS tide models are now used without needing to use the lib/iers/ source, and the functions used are bundled into src/tides.c That leaves only the tropospheric mapping being modified by the IERS_MODEL compilation option - to use GMF from the IERS library.

AndreHauschild, 2025-04-04:
Understood, thank you! So that means that 

>  if(IERS_MODEL)
   target_link_libraries(-liers -lgfortran)
endif()

should be removed from `app/consapp/CMakeLists.txt`.

JensReimann, 2025-04-04:
No matter if libiers is hardly used, I added support to cmake to compile with this library.