rtklibexplorer/RTKLIB #567: Inconsistency of PHASE SHIFT Option in rtkconv/convbin

liusheng2020, 2025-01-21:
The GUI application rtkconv includes a PHASE SHIFT option, while the CUI application convbin seems to lack this option (at least I couldn't find it in the consapp\convbin\convbin.c file or by running convbin -?). This causes discrepancies in the conversion results.

[image]

For example, when GPS L2X PHASE SHIFT = -0.25000, all GPS L2X PHASE observation value elements in the conversion results exhibit a -0.25 difference.

[image]

[image]

Is this a bug, or do I need to add a specific option to make the conversion results of convbin consistent with those of rtkconv?

ourairquality, 2025-01-21:
My reading of the RINEX spec is that the observations should have aligned phases, aligned to their reference signals, and that the header phase shifts are for recovering the original unaligned signals. This option hard codes some phase shifts, those in the RINEX spec table A23, which appear to be for recovering the unaligned phases. It's does seem odd that the observations change based on the header phase shifts. Would have expected the internal rtklib observations to be the aligned phases rather than attempting to recover the unaligned, and that the observations are always the aligned phases so should not change based on the header phase shifts?

If the phases from the ublox f9 raw data are already phase aligned (even if after some decoding adjustments) then it would appear that the RINEX header phase shift should be zero, that this conversion option should not be used. If these are not phase aligned, and if the internal observation phases were intended to be the original unaligned phases, then they would need to be output in the RINEX header and the values adjust to be aligned.

But still, perhaps the application of the phase shift is not correct. Are the internal observation phases intended to be phase aligned, or not??

rtklibexplorer, 2025-01-21:
Regarding the difference between the GUI and the CUI, it does look like the CUI is missing support for this feature.  

If you want to fix this yourself before the next release and you are able to modify and build the code, I think it simply needs this code:

        else if (!strcmp(argv[i],"-phshift")) {
            opt->phshift=1;
        }

added right after this code in convbin.c:

        else if (!strcmp(argv[i],"-halfc")) {
            opt->halfcyc=1;
        }

Then it can be enabled by adding "-phshift" as an option to the command line.

I haven't tried this but believe it should work.