rtklibexplorer/RTKLIB #568: Request: Add support for specifying antenna dome type in `convbin` and `rtkconv`

millerjs2, 2025-01-28:
Neither `convbin` nor `rtkconv` properly format the `ANT # / TYPE` line in the RINEX header of their output observation files. For some reason, the RINEX specification documentation isn't clear on this, but the `ANT # / TYPE` line really should be thought of as `ANT # / TYPE / DOME` - but the dome type is lumped in with the antenna type. The dome specification must be formatted correctly, however, otherwise PPP services (e.g. opus and CSRS-PPP) will not interpret the antenna entry correctly and will fail to make those corrections. 

The correct formatting for the `ANT # / TYPE` line in the header is as follows:

- records 1-20 are the serial number
- records 21-36 are for the antenna type
- records 37-40 are for the antenna dome

Currently, RINEX observation files must be manually edited to include this after processing. 

I suspect this has gone unaddressed for so long because few RTKLIB users are using PPP services with calibrated antennas? Either way, it seems like it would be an easy fix, and would be a significant usability improvement.

ourairquality, 2025-01-29:
The type / dome is currently handled as a single 20 character string. E.g. convbin -ha "4539433521/TRM59800.00     NONE" ... Same for the config file e.g. 'ant2-anttype=TRM59800.00     NONE'  It is currently up to the user to correctly format this pair, 
 but that is just a copy a paste matter. The third value in the convbin -ha argument is an antenna setup id, which does not appear to be used by rtklib.

millerjs2, 2025-01-29:
Handling the type / dome as a single 20 character string is incorrect, though, and PPP services will incorrectly process (or outright reject) RINEX observation files that are formatted this way. The dome must be defined on the 37th to 40th columns of the `ANT # / TYPE` row of the header. 

I did test and can confirm that in `rtkconv` it is possible to manually place the correct number of spaces between the antenna type and dome type, but that's a clunky workflow. If you're only ever using one antenna then you can save it in the ini file, but if you're processing data from different sites with different antennas, you have to have the requisite number of spaces memorized or saved elsewhere, or else manually edit the obs file to align the text. This is especially frustrating as `rtkconv` has a field in the options box right next to the antenna type field which could be used for the dome type. I've found similar for `convbin`.

Ultimately I would argue that the RINEX standard using this kind of formatting is obscene, but it is a standard, and other software outputs correctly formatted files.

rtklibexplorer, 2025-02-04:
Fixed with PR #569

closed 2025-02-04 by rtklibexplorer