rtklibexplorer/RTKLIB #585: How to compile rtklib demo5 with visual studio

changjiy, 2025-03-01:
[image]

[image]
How to solve it? Better make a tutorial

JensReimann, 2025-03-01:
How did you set up the project in VS? I'm not an expert in VS, but the comnav support is broken. This file (comnav.c) should not (and does not need to) be compiled.

changjiy, 2025-03-01:
Delete comnav.c from the source file？

[image]
Two other mistakes：

[image]
Is this line of code commented

changjiy, 2025-03-01:
[image]

JensReimann, 2025-03-01:
The same applies to the tesus receiver support. It should not be included in the compilation if you use one of the predefined compilation methods.
The receiver support is still in the code base waiting for a volunteer to fix the code.

changjiy, 2025-03-01:
Thanks,my God!