rtklibexplorer/RTKLIB #590: RTKCONV (GUI) does not convert Beidou L5 signals (B2)

alex-skyfora, 2025-03-26:
I am trying to convert RTCM to RINEX. My NEO-F9P RTCM log contains GPS, Galileo and Beidou data on L1 + L5. I am converting the .rtcm log to RINEX with RTKCONV with these options:

[image]

But the resulting RINEX file does not contain B2:

[image]

I have tried with RINEX TOOLS and it shows the B2 signals correctly:

[image]

I haven't found anyone else mentioning this, so I created this issue.

ourairquality, 2025-03-26:
Does it work using the command line tool?

alex-skyfora, 2025-03-26:
do you have a command I could try out?

ourairquality, 2025-03-27:
`convbin -os -od -tr 2025/03/26 12:00:00 rover.rtcm3`

alex-skyfora, 2025-03-27:
Thanks a lot for the answer!

Unfortunately the CLI version produces the same result

alex-skyfora, 2025-03-28:
Absolutely, here you go:
[log_2025086.zip](https://github.com/user-attachments/files/19500917/log_2025086.zip)

The `.rtcm3` file is the RTCM log file, the `.obs` file is the output of RTKCONV with missing BeiDou signals, and `.25o` file is the output from RINEX TOOLS free online conversion tool which does it correctly.

ourairquality, 2025-03-28:
Thank you. This might have been fixed already. Here is the output using the current development demo5 code, and the L5 code is included.

`convbin -os -od -tr 2025/03/26 12:00:00 L00702FIN0_113852.rtcm3`

```
CONVBIN demo5 b34k                      20250328 110904 UTC PGM / RUN BY / DATE 
format: RTCM 3                                              COMMENT             
log: L00702FIN0_113852.rtcm3                                COMMENT             
...
G    8 C1C L1C D1C S1C C5Q L5Q D5Q S5Q                      SYS / # / OBS TYPES 
E    8 C1C L1C D1C S1C C5X L5X D5X S5X                      SYS / # / OBS TYPES 
C   12 C2I L2I D2I S2I C1P L1P D1P S1P C5P L5P D5P S5P      SYS / # / OBS TYPES 
...
> 2025 03 27 11 38 52.0000000  0 30                     
G05  21935027.320   115268594.9101        612.585          45.438                                                                  
G09  24736830.279   129992211.8651      -1310.102          39.500    24736845.094    97072157.6621       -978.335          43.000  
G16  21878181.689                3       3096.841          44.188                                                                  
G18  21204829.193   111431917.3591       2515.144          46.625    21204844.253    83212550.5441       1878.193          46.500  
G23  25161916.586   132226525.0121       4349.687          33.313    25161939.707    98740372.0451       3248.156          33.188  
G25  24759194.926   130110599.0171      -3553.663          36.313    24759215.814    97160680.1541      -2653.699          36.063  
G26  20966890.950   110181485.3341        943.927          47.438    20966903.114    82278827.8791        704.877          46.938  
G28  25143953.329   132131622.5351      -3153.656          35.125    25143977.550    98670314.9861      -2354.971          43.188  
G29  21078921.255   110770515.0101      -1507.310          48.063                                                                  
E07  27527737.159   144658760.8111       1037.691          39.313    27527752.515   108024627.6381        774.919          36.000  
E08                                                                  28099372.747   110267583.7281       -999.357          42.250  
E10  24204707.118   127195960.1471       2247.510          47.063    24204719.036    94984403.4941       1678.340          44.813  
E11  26288832.345   138149241.8281       3346.067          39.250    26288843.583   103163285.7821       2498.680          38.000  
E12  23705656.539   124573760.9731        860.878          41.625    23705667.288    93026531.0511        642.865          43.438  
E16  23529398.641   123647498.8641        251.047          47.063    23529410.978    92334318.2651        187.477          46.813  
E24  24463885.543   128559275.9521      -1252.791          44.375    24463899.747    96001330.4761       -935.523          43.813  
E25  24109789.047   126698241.7171       1691.516          47.000    24109802.327    94611634.0351       1263.155          43.250  
E31  27784282.048   146007444.8131      -2698.693          36.375    27784302.483   109031214.8151      -2015.267          38.000  
E33  25702456.538   135066766.1081      -1968.282          45.438    25702469.504   100861567.2141      -1469.811          44.438  
C19                                                                  25020151.427   131481369.2461        983.693          43.063    25020166.364    98184109.6791        734.587          42.563  
C20                                                                  26340475.030   138419721.3921      -1526.839          39.250    26340486.844   103365695.1751      -1140.161          41.125  
C22                                                                  25488972.526   133946133.3191       3433.644          37.563    25488985.770   100024807.8131       2564.089          40.375  
C23                                                                  24737829.837   129997725.1031       2610.047          42.688    24737839.104    97076082.3251       1949.050          44.313  
C24                                                                  24733332.414   129974949.8231      -2884.834          38.438    24733344.245    97058996.7161      -2154.233          40.375  
C25                                                                  22678909.822   119178613.6911       -473.541          44.938    22678920.920    88997252.7691       -353.621          43.313  
C34  22250839.463   115866319.8391       2454.354          46.750    22250835.415   116928590.3381       2476.867          42.375    22250847.909    87317341.2251       1849.609          43.313  
C35                                                                  25154716.643   132188855.3821      -2641.988          40.688    25154733.203    98712523.7171      -1972.905          39.750  
C38                                                                  37598749.056   197582139.9651        422.165          44.563    37598760.486   147545633.7461        315.264          44.375  
C43                                                                                                                                  26322952.111   103297393.0871       2856.916          34.313  
C44  21706785.850   113032935.0231       -784.120          49.438    21706780.929   114069472.4351       -791.321          46.438    21706794.732    85182436.3921       -590.906          45.438  
```

alex-skyfora, 2025-04-01:
That's great! Thanks for looking into this!

Do you have any estimate of when this will make it into the next release?

rtklibexplorer, 2025-04-18:
I hope to get a new release out in the next couple of weeks

closed 2025-04-18 by rtklibexplorer