rtklibexplorer/RTKLIB #591: Map view not woking

EmmanuelMess, 2025-03-30:
Version b34k

[image]

JensReimann, 2025-03-31:
The map view is not available as the required web browser component (QWebEngine; part of Qt) was not found during compilation. If you are using cmake to compile, please try the latest git version as the commit "Qt GUI: improve CMake support" improves the detection of this component.
Please report if the issues remains.

EmmanuelMess, 2025-03-31:
> The map view is not available as the required web browser component (QWebEngine; part of Qt) was not found during compilation. If you are using cmake to compile, please try the latest git version as the commit "Qt GUI: improve CMake support" improves the detection of this component.

I am using Qmake in a docker:

```
FROM ubuntu:noble

RUN apt-get update && \
    apt-get install -y \
        git \
        build-essential \
        libpng-dev \
        qtbase5-dev \
        qt5-qmake \
        libqt5serialport5-dev \
        qt5-qmake-bin && \
    rm -rf /var/lib/apt/lists/*

WORKDIR /tmp
RUN git clone -b b34k https://github.com/rtklibexplorer/RTKLIB.git

# First CUI
WORKDIR /tmp/RTKLIB/app/consapp/
RUN make -j

# QT
WORKDIR /tmp/RTKLIB/app/qtapp
RUN qmake
RUN make -j
RUN ./install_qtapp
```

EmmanuelMess, 2025-04-01:
`qt5-webengine-dev` does not exist, `libqt5webengine5` does exist but its not fixing the issue. Do you have a list of dependencies? Can you add them to the readme?

EmmanuelMess, 2025-04-01:
`qt5-webengine-dev` does not exist, and installing `qtwebengine5-dev` or `libqt5webengine5` doens't work.
Could you provide a list of dependencies in the readme?