rtklibexplorer/RTKLIB #594: A Drag-and-Drop Issue in rtkplot_qt on Windows and Linux(Ubuntu)

liusheng2020, 2025-03-31:
I have encountered an issue with the drag-and-drop functionality in rtkplot_qt from demo5.
On Windows, dragging a trajectory file (*.pos) into the rtkplot_qt.exe window has no effect; the drop action does not work. In contrast, the Embarcadero version of RTKPLOT on Windows supports drag-and-drop properly, which greatly improves efficiency by eliminating the need to manually open file paths.
On Ubuntu,  drag-and-drop only works for solution 1, but does not work for solution 2 (it only replaces the trajectory display of solution 1 instead of loading solution 2 correctly).
It would be great if this functionality could be improved.

See GIF demo bellow:
Windows:

[image]

Ubuntu:

https://github.com/user-attachments/assets/5d9df225-c48a-45be-b601-977c7b8bc285

[image]

JensReimann, 2025-03-31:
Thanks for reporting the bug. It should be fixed with the latest "Qt GUI: rtkplot_qt" commit.

JensReimann, 2025-04-01:
Please check again if compilation works again

liusheng2020, 2025-04-01:
@JensReimann  Thank you for your response! However, there is still an error preventing the project from being fully compiled with CMake. The error is as follows:
`
[build] [ 21%] Building C object src/CMakeFiles/rtklib.dir/rcv/unicore.c.o
[build] [ 21%] Linking C shared library /media/ls2410/608C02658C023656/IONcomp/rtilib_collection/demo5VscodeCmakeQTandCmd/lib/librtklib.so
[build] /usr/bin/ld: CMakeFiles/rtklib.dir/sofa.c.o: in function `epv00':
[build] sofa.c:(.text+0x0): multiple definition of `epv00'; CMakeFiles/rtklib.dir/rtkcmn.c.o:rtkcmn.c:(.text+0xc620): first defined here
[build] /usr/bin/ld: CMakeFiles/rtklib.dir/sofa.c.o: in function `moon98':
[build] sofa.c:(.text+0x7a0): multiple definition of `moon98'; CMakeFiles/rtklib.dir/rtkcmn.c.o:rtkcmn.c:(.text+0xc630): first defined here
[build] collect2: error: ld returned 1 exit status
[build] make[2]: *** [src/CMakeFiles/rtklib.dir/build.make:785：/media/ls2410/608C02658C023656/IONcomp/rtilib_collection/demo5VscodeCmakeQTandCmd/lib/librtklib.so] 错误 1
[build] make[1]: *** [CMakeFiles/Makefile2:333：src/CMakeFiles/rtklib.dir/all] 错误 2
[build] make: *** [Makefile:156：all] 错误 2
[proc] 命令“/usr/local/bin/cmake --build /media/ls2410/608C02658C023656/IONcomp/rtilib_collection/demo5VscodeCmakeQTandCmd/build --config Release --target all -j 3 --”已退出，代码为 2
[driver] 生成完毕: 00:00:53.417
[build] 生成已完成，退出代码为 2`

[image]

ourairquality, 2025-04-01:
Try the fix in https://github.com/rtklibexplorer/RTKLIB/pull/595

closed 2025-04-06 by JensReimann