rtklibexplorer/RTKLIB #632: rtknavi_qt freezes when a marker is added

AndreHauschild, 2025-04-25:
The Qt app `rtknavi_qt` freezes when a marker is added via `Mark...`. The system and compiler versions are:

- Ubuntu 24.04.2 LTS
- gcc version 13.3.0
- cmake version 3.28.3
- Qt version 5.15.13

JensReimann, 2025-04-26:
I was able to reproduce the issue. In my case, there were no data coming on the input causing rtklib to stay in the reading function holding the global rtksvr lock.
Setting a marker calls rtksvrmark() with also tries to acquire the lock causing the freeze of the GUI.

Can you please check if in your case, rtknavi_qt also stuck reading data?

AndreHauschild, 2025-04-28:
Yes, this is exactly what I am seeing as well.

ourairquality, 2025-05-19:
Could you retest this now after https://github.com/rtklibexplorer/RTKLIB/pull/645 which might have resolved this.

AndreHauschild, 2025-05-19:
Excellent, that solved it. Thanks for your help!

closed 2025-05-19 by AndreHauschild