rtklibexplorer/RTKLIB #653: Please decide for a proper name / version number

stoecker, 2025-05-07:
I'm building rtklib for openSUSE Linux distribution and recently added your variant as well (the repo also contains two beta versions and the last official release).

https://build.opensuse.org/package/show/Application:Geo/rtklib-explorer

Could you please choose a proper name and version for your package?

"demo5 b34k" sounds strange. Especially as it implies the 2.4.3 as well.

For openSUSE I did choose 2.4.3.b34k, but that differs from the release versions as it leaves out the demo5.

My proposal would be to simply start with "rtklib 3.0.0" or even "rtklib 3.0" and drop all the mumbo-jumbo. It seems your repo did take over development after Takasu left, so I think version number can start fresh.

rtklibexplorer, 2025-05-10:
I think you make a good point.  The repo has evolved considerably since it started as a series of simple demonstrations for my early blog and it's probably time to reflect that in the naming and versioning, especially since the main fork has been fairly inactive for a while.  However, I don't want to choose a path that impedes any future work on the main fork either from Takasu or potentially a new owner, so I don't think RTKLIB 3.0 is the right choice.  I would lean more towards something like RTKLIB-EX (for extended), or RTKLIB-LC (for low cost), or even RTKLIB-D5 (to keep the connection to demo5) with an initial version of 1.0.0.  At that point it may make sense to start a new fork.  If I did that, I would probably reformat the code to a more modern style as well, as "Our Air Quality" has previously suggested in another issue.  Anyone else have thoughts on these suggestions or would like to propose other options?

tonycanike, 2025-05-10:
Agreed it's time for a simpler name.  

Also agreed on using a naming variant like RTKLIB-EX to differentiate from the original RTKLIB. 

So many reasons, including basic courtesy and respect, to use a variation on the RTKLIB name. 

The "RTKLIB-EX" name could indicate "extended" and/or "explorer", so I like the "RTKLIB-EX" name.

stoecker, 2025-05-10:
Rtklib-Ex sounds good. I'd still propose to not lower the version. Either start with 2.5 or 3.0.

rtklibexplorer, 2025-05-24:
I've just released demo5 B34L which will be the last version with the demo5 name.  I will most likely switch to RTKLIB-EX 2.5.0 for the next release but am still open to suggestions.  I also haven't decided if this will be a new repo or just a name change in the existing repo. 

Regarding support for higher-end receivers, my focus has always been on low-cost receivers.  When I started this fork, almost all low-cost receivers were single-frequency.   We are now seeing low cost receivers with 4+ frequencies making the differences between low-end and high-end much less significant.  Also, I believe the very welcome recent activity from several other contributors has broadened the scope to include better support for higher-end receivers.

stoecker, 2025-05-25:
> If new, then with initial version 1.0.0 e.g. RTKLIB-EX 1.0.0. Using the number 2.5.0 would suggest that RTKLIB-EX had earlier versions.

Starting with a lower version is a bad idea. Beside the fact that it will result in a lot of troubles for packagers like me who create RPMs or DEB files, a lower version will give the impression this software is less than current rtklib where its the other way around.

Also why should "suggest that RTKLIB-EX had earlier versions" be bad? It's the truth, there are earlier versions.

> It is a fork of RTKLIB and unless the license conditions require it, there is no obligation to keep the RTKLIB name. You can start with any name, and in the program description mention that it is a continuation of RTKLIB.

A totally new name is likewise a bad idea. First that means that you also need to rename the tools which breaks compatibility without a good reason and second it means nobody will understand that it is a followup to rtklib and thus it will loose what this repository here achieved: continuing the development.

ourairquality, 2025-05-27:
The PPP solution in RTKLIB does not appear to have evolved with 'growing maturity'. Most of the work is just on the plumbing of data flow and presentation, regular programming grind. A lot of exacting mathematics would be involved in expanding to multi-frequency PPP, and even then it appears to be a matter for research, just how the phase centers vary at the satellites for particular generations of satellites and signals etc, advanced models for the propagation delays etc. If it were mature then all these inputs would be in open standard products to use. There are also suppliers using proprietary protocols in this area, I guess to try to gain some vertical market barriers, and that does not appear to be open and not something we are likely to gain access to. We don't even appear to have representatives on RTCM forums, even some of the top supplies are not responsive to issues we find  and clarification support we need. It is likely a tough gig. The receivers we use are somewhat a black box to us. We do what we can.

Yes, there is a lot of work in the PPP area, but it appears to still be a compromise compared with RTK, still talking about decimetre level accuracy, still with long initialisation times of 20 to 30 minutes. PPP is marketed as not needing an observation data stream, but communication are certainly improving and it is becoming even more practical to have a real time observation data stream. Once we have RTKLIB running in the web browser and thus on mobile devices it will typically be trivial to have an observation data stream. Who would compromise using PPP for post processing when there is PPK. If you need quick real time on-site results wouldn't you use RTK, and if you need the best post processed results wouldn't you use PPK. If you are trying to get the best results, or even a usable result, in challenging environments wouldn't you use PPK.

I suspect that the mathematics involved in RTK solutions is a bit more robust, the double difference helping, and so more practical for us to develop, and the published methods appear more developed. PPP is going to needed a calibrated antenna for best results, but at the cheap end two matched uncalibrated antennas might be an alternative, hoping that the PCO and PCV cancel. Even if there is not a static base near a site then it may well be practical to set up a base within a few kms of a site and either on a known marker or just on a stable point for relative measurements, and isn't that better than PPP.

For PPP there is competition from other open source projects such as Pride-PPP https://github.com/PrideLab/PRIDE-PPPAR and Ginan https://github.com/GeoscienceAustralia/ginan and PPP development in MADOCA-PPP https://github.com/JAXA-SNU/MALIB https://github.com/QZSS-Strategy-Office/madocalib and wouldn't people just use one of those tools for PPP. Even these projects do not appear to have a lot of open source developers contributing, some not very responsive to issues either, probably also a though gig to get funding. Do people know of other relevant projects or work on PPP?

These are some of the reasons that my focus is on the RTK processing, and maintenance of the data flow paths in RTKLIB, and why RTK and PPK seem to have an ongoing niche. Am I mistaken??

rtklibexplorer, 2025-07-30:
The RTKLIB-EX 2.5.0 is now the latest full release and the demo5 branch readme has been updated to indicate it is retired.