rtklibexplorer/RTKLIB #719: Can't parse MSM3 messages

juffe84, 2025-07-12:
Hello.
 I have one stream what I can't parse with rtklib.
Snip show like this.
[image]

RTKNAVI and strsvr cant parse it. What is problem?

[MSM3.txt](https://github.com/user-attachments/files/21199677/MSM3.txt)

Here is also RAW file from strsvr. Couple of seconds.

juffe84, 2025-07-13:
No. I can't change message types. Too many user with MSM3 in this stream. 😁 

Only what I need to do is convert this stream to legacy rtcm3.0 that I can use also Trimble R8.

ourairquality, 2025-07-13:
The later trimble R8 models might support MSM7. Are you wanting to receive a MSM3 stream for a RTKLIB based rover, and where the available base streams are MSM3? MSM3 is limited, modulo 1ms ranges, needs clock steering, and clocks need to be close, and a lot of lower end receivers might not be suitable. If just wishing to pass through the MSM3 data then can it just be handled as raw data without parsing it.

ourairquality, 2025-07-13:
Btw RTKLIB does have MSM1,2,3 encoding support, but still not sure how well that would work with the low end receivers that jump the clock.

juffe84, 2025-07-13:
I want to use my R8 on the place where is only MSM3 signal. MSM3 signal is coming from high end receiver. But what we need that we can get decode_msm3 function? Is it complicated to code?

ourairquality, 2025-07-13:
The complexity is filling in the missing rough range, as MSM1,2,3 have only the modulo 1ms range and don't encode enough data to decode a complete range, so it would need to compute the satellite ranges from the base position and satellite ephemeris.